stephendade/Rpanion-server #9: [Feature Request] Allow (multible) and different sources

stundenblume, 2020-06-30:
If it would be possible to have different sources for the mavlink stream you could use it as "mavlink broker".
Scenario:
1. drone with "Rpanion-server" is connected to "Rpanion-server(broker)". long range via VPN, short range via Wifi.
2. Different Devices (PC/Tablet/...) with GCS are connected to "Rpanion-server(broker)"

This is my current setup which I have realised with mavlinkproxy, without your nice GUI.

stephendade, 2020-07-01:
I'm not 100% sure of what you mean. Are you saying that Rpanion-server is not connected to the flight controller via serial port?

stundenblume, 2020-07-01:
Yes. I would like to use your Rpanion-server as "Rpanion-Broker" on the ground.

I'm sorry if this feature request is to far from your idea what Rpanion-server should be.

Scenario:
A first Raspberry with Rpanion-server in the drone, connected to an LTE router.

A second Raspberry with "Rpanion-server" on the ground. Connected to the first via LTE (VPN). If possible, a second parallel connection via WiFi.

To the second Raspberry on the ground I connect different devices (Tablet, PC, ...)

I run this setup currently with MAVproxy.

On the ground and up to 500m the stable and fast WiFi connection is in use and further away it will switch to LTE.

I home this makes sense...

ele-dev, 2021-02-07:
That sounds like a very interesting idea.
I would also be interested in a solid solution to switch seamlessly between wifi connection and LTE over VPN connection, depending on how far the drone moves away from the Ground Station, because LTE traffic consumes the limited data volume per month and is only required for long distance operations.

stephendade, 2021-02-08:
For multiple inputs on the ground, I'd recommend the MAVProxy GCS. It can act as a router and input/output to many sources.