stephendade/Rpanion-server #11: Installation Dependencies deprecated and wireless interfaces down

hartungalex24, 2020-08-25:
Hello, 

I was very excited to see your nice react web UI interface for interfacing with mavlink and flight controller. However, during installation I have run into a few problems. 

I am using a Raspberry Pi 4 with Pixhawk 4 and followed the installation instructions at: https://github.com/stephendade/Rpanion-server

In my case, I opted for the manual installation because I already had many programs configured for the computer vision side of things. However, I have been unable to compile mavlink-router from source. It has thrown errors saying my GCC C compiler is not functioning. However, I have compiled and built many other programs before this just fine. Perhaps something specifically wrong with mavlink-router current build? 

In hope that I could come back to this, I ran 'npm install' in the rpanion-server directory to get all the main packages for the UI. Unfortunately, many of these packages are deprecated and I was left with a failed installation. 

Lastly, when I opted to not continue down this path anymore and rebooted my PI, it is now booting into pi@rpanion and my Wireless interface is no longer found and cannot connect to internet. 

Do you have any suggestions / more information I can provide? 

Thanks in advance,
Alex

stephendade, 2020-08-26:
Can you post the error messages?

hartungalex24, 2020-08-26:
@stephendade Thank you for your time. I greatly appreciate any help. Please find my error messages below and if you need me to run anything additional I will be very prompt with my response :)

Here is my dmesg output focused on the failing driver it seems:
pi@rpanion:~ $ dmesg | grep brcm
[    0.927362] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.927384] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.927448] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    0.927528] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00bfffffff -> 0x0000000000
[    0.983011] brcm-pcie fd500000.pcie: link up, 5 GT/s x1 (SSC)
[    0.983327] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    6.414811] brcmfmac: F1 signature read @0x18000000=0x15264345
[    6.426268] brcmfmac: brcmf_fw_alloc_request: using b**rcm/brcmfmac43455-sdio for chip BCM4345/6**
[    6.426713] usbcore: registered new interface driver brcmfmac
[    6.448094] **brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2**
[    6.689766] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.707616] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
[   10.299058] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.824855] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

[dmesg_brcm.txt](https://github.com/stephendade/Rpanion-server/files/5130517/dmesg_brcm.txt)

I thought maybe I need to figure out how to point the firmware to load to the right path. And found this https://github.com/RPi-Distro/firmware-nonfree/blob/master/brcm/brcmfmac43430-sdio.raspberrypi-rpi.txt which seems to be the same driver. Just not sure what exactly to do here.  

The full log can be found here:
[DMSG.txt](https://github.com/stephendade/Rpanion-server/files/5130518/DMSG.txt)

The major issue is my Wireless Network Interface 'cannot be found' in terms of the GUI and shows a 'dhcpd connection lost'. I ran your Raspi-2-3-4.sh deploy script to get things properly configured. But I am worried at some point this changed the network configuration settings. 

Here are some behaviors related to the WIFI driver. 

I can run 
`sudo iwlist wlan0 scan` and get results back. So my network interface is able to scan the network (sometimes, other times it says Interface doesn't support scanning)

`sudo rfkill list` shows:
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

Then I try to connect to my network by:
`sudo wpa_supplicant -iwlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf`
OUTPUT: 
Successfully initialized wpa_supplicant
ctrl_iface exists and seems to be in use - cannot override it
Delete '/var/run/wpa_supplicant/wlan0' manually if it is not used anymore
Failed to initialize control interface 'DIR=/var/run/wpa_supplicant GROUP=netdev'.
You may have another wpa_supplicant process already running or the file was
left by an unclean termination of wpa_supplicant in which case you will need
to manually remove this file before starting wpa_supplicant again.

nl80211: deinit ifname=wlan0 disabled_11b_rates=0

I tried to check for open files or any other wlan0 and pcp and delete them. But to no success. 

`sudo lsof | grep wlan0`
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
wpa_suppl  390                            root   15u     unix 0x0b7dda44       0t0      18353 /run/wpa_supplicant/wlan0 type=DGRAM
wpa_suppl  390                            root   19u     unix 0xcd7a3a16       0t0      18358 /run/wpa_supplicant/p2p-dev-wlan0 type=DGRAM

In regards to the npm installation issues, I retried the command today and it didn't crash. I attached the output here: 
[npm_package_installation.log](https://github.com/stephendade/Rpanion-server/files/5130569/npm_package_installation.log)

Thanks!

stephendade, 2020-08-27:
>The major issue is my Wireless Network Interface 'cannot be found' in terms of the GUI and shows a 'dhcpd connection lost'. I ran your Raspi-2-3-4.sh deploy script to get things properly configured. But I am worried at some point this changed the network configuration settings.

The installation script changes how networks are managed on the Pi. It replaces the default network manager with ``nmcli``.

``wpa_supplicant`` is disabled as prt of the installation and won't be able to be used.

hartungalex24, 2020-08-27:
HI @stephendade , thanks for pointing this out. I have configured the settings and successfully connected back to the internet using nmcli. 

This makes a lot more sense now. I had been calling wpa_supplicant.conf but it seems that dhcp was no longer assigning an IP address. I did manage to manually assign an IP address and use wpa_supplicant to re-connect to the internet. But now I am quite interested in nmcli. 

My ultimate goal is to have this PI be a mobile companion computer for my Rover/Copter set-up. However, I still want to be able to switch between accessing my local internet at home for debugging and broadcasting an access point when I am in mobile mode. 

If I create a service for this access point which boots on start, will I still be able to connect to my local access point? If not, perhaps I can call the hotspot script for my other main program on start-up during missions. 

Either way, 
Thanks for your help and I will edit this post after successfully getting your server running.

hartungalex24, 2020-08-28:
Everything is working as expected. Looking forward to bringing computer vision into the mix. Thanks for your help

closed 2020-08-28 by hartungalex24