stephendade/Rpanion-server #12: how to swarm

jiaozhuliu, 2020-09-16:
I want swarm drone, how connet long distance。

stephendade, 2020-09-20:
Most people use 4G LTE modems for long distance communications.

You would also need to run a VPN to network the drone and ground station. I've used Zerotier in the past with good results.

jiaozhuliu, 2021-03-03:
Can use rpanion to swarm？ rpanion connect wifi hotspot

stephendade, 2021-03-04:
Yes, you can connect multiple rpanion-server instances to a single Wifi AP for swarming communications.

jiaozhuliu, 2021-03-08:
I used the IP output to the computer to achieve the connection, but UPD or TCP is better for bee swarm

TonyStark2007, 2021-06-07:
Hi, How can I use two rpanion server (connected to 2 pixhawk) in mission planner or QGC? Intention is not swarming but using multiple drones from single GCS. Is it possible?

stephendade, 2021-06-07:
> How can I use two rpanion server (connected to 2 pixhawk) in mission planner or QGC? Intention is not swarming but using multiple drones from single GCS. Is it possible?

Both Mission Planner and QGC can connect to multiple vehicles. You will need to check their documentation as to the instructions.

lida2003, 2024-05-28:
As wifi has been used for AP, bluetooth might be avaliable for AdruRemoteID (Bluetooth 4 Legacy Advertising or Bluetooth 5 Long Range + Extended Advertising).

1. Raspberry Pi ZeroW Bluetooth 4.1, BLE
2. Raspberry Pi3B+ Bluetooth 4.2, BLE
3. Raspberry Pi4B Bluetooth 5.0, BLE

@stephendade Is there any plan to implement Remote ID support on Rpanion?

- [Remote ID (aka Drone ID)](https://ardupilot.org/copter/docs/common-remoteid.html)
- [OpenDroneID](https://ardupilot.org/dev/docs/opendroneid.html#opendroneid)
- [ArduRemoteID](https://github.com/ArduPilot/ArduRemoteID)

stephendade, 2024-05-29:
No plan to support Remote ID

closed 2024-10-10 by stephendade