stephendade/Rpanion-server #13: [Eachine ROTG01] UVC usb input incompatibility

joaomruas, 2020-09-23:
Currently using server as man in the middle with telemetry radio and video receiver connected via usb.

Having problems with this device: Eachine ROTG01 UVC
No problems with common usb camera.

USB device is detected and can be selected on video streaming configuration.
No resolution options are shown.
Once video streaming is enabled a blank web page is presented (deviceIP:WebPort/video) and upon refresh streaming was not enabled. 

Any idea what to look for?

Using Arch Linux:
    Can open video (/dev/video) stream on VLC:
        Codec: Motion JPEG Video (MJPG)
        Video resolution: 640x480
        Frame rate: 60
        Decoded format: Planar 4:2:2 YUV full scale
        Color space: ITU-R BT.601 Range
    Can identify USB configuration reporting "UVC non compliance - GET_DEF(PROBE) not supported":
        dmesg:
                [   87.663156] usb 2-3: new high-speed USB device number 2 using xhci_hcd
                [   87.982784] usb 2-3: New USB device found, idVendor=18ec, idProduct=5850, bcdDevice= 1.00
                [   87.982793] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
                [   87.982797] usb 2-3: Product: USB2.0 PC CAMERA
                [   87.982801] usb 2-3: Manufacturer: ARKMICRO
                [   87.984270] uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (18ec:5850)
                [   87.993717] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
                [   87.994544] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
                [   87.994556] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
                [   87.994918] input: USB2.0 PC CAMERA as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/input/input23
                [   88.695167] usb 2-3: 3:1: cannot get freq at ep 0x82
                [   88.699427] usb 2-3: Warning! Unlikely big volume range (=11903), cval->res is probably wrong.
                [   88.699443] usb 2-3: [2] FU [Mic Capture Volume] ch = 2, val = 384/12287/1
                [   88.699901] usbcore: registered new interface driver snd-usb-audio
                [   88.776906] usb 2-3: 3:1: cannot get freq at ep 0x82
                [   88.781502] usb 2-3: 3:1: cannot get freq at ep 0x82

stephendade, 2020-09-24:
Can you run the ``gst-device-monitor-1.0`` command and post the output? That will tell me how gstreamer is detecting the camera.

joaomruas, 2020-09-25:
Yes, no problem, here is the requested output:

```
pi@rpanion:~ $ gst-device-monitor-1.0
Probing devices...
Device found:
	name  : USB2.0 PC CAMERA
	class : Video/Source
	caps  : image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1 };
	        image/jpeg, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1 };
	properties:
		udev-probed = true
		device.bus_path = platform-3f980000.usb-usb-0:1.2:1.0
		sysfs.path = /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video0
		device.bus = usb
		device.subsystem = video4linux
		device.vendor.id = 18ec
		device.vendor.name = ARKMICRO
		device.product.id = 5850
		device.product.name = "USB2.0\ PC\ CAMERA"
		device.serial = ARKMICRO_USB2.0_PC_CAMERA
		device.capabilities = :capture:
		device.api = v4l2
		device.path = /dev/video0
		v4l2.device.driver = uvcvideo
		v4l2.device.card = "USB2.0\ PC\ CAMERA"
		v4l2.device.bus_info = usb-3f980000.usb-1.2
		v4l2.device.version = 267126 (0x00041376)
		v4l2.device.capabilities = 2225078273 (0x84a00001)
		v4l2.device.device_caps = 69206017 (0x04200001)
	gst-launch-1.0 v4l2src ! ...

´´´

stephendade, 2020-09-26:
I've updated Rpanion-server to support that camera. Give it a test and let me know if it works.

joaomruas, 2020-09-26:
Thanks, saw commit on Master, will be testing tomorrow on clean install.

joaomruas, 2020-09-27:
Working correctly
Thanks

closed 2020-09-27 by joaomruas