stephendade/Rpanion-server #22: Add UDP Video Stream to support / qgroundcontrol

dannyb131, 2021-01-01:
could you add udp video stream to support qgroundcontrol.
in my testing udp reduces the latency by upto 100ms that the currently only available option in qgroundcontrol is rtsp that works with Rapnion.

[https://docs.qgroundcontrol.com/master/en/SettingsView/General.html#video](url)
[https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md](url)

Thanks

stephendade, 2021-01-02:
Thanks for investigating! I'll add it in the coming weeks.

stephendade, 2021-01-08:
I've got the feature working in #24 and is working for me. Are you able to test too?

dannyb131, 2021-01-13:
Hi Stephen,

seems to be working! good stuff!

Thanks

stephendade, 2021-01-14:
Cool, thanks for testing! I'll merge it in

closed 2021-01-14 by stephendade