stephendade/Rpanion-server #27: Getting a nmcli to work with an USB dongle. [Not directly related to rpanion]

Drkstr, 2021-05-08:
I am using the Huawei e8372 USB WiFi dongle to connect to the internet. It worked fine till I installed rpanion server. 

I know rpanion replaces the default network manager with `nmcli`. 

Would you happen to know how I could get `nmcli` to use a USB WiFi dongle to connect to the internet?

stephendade, 2021-05-09:
What output do you get if you enter the ``ifconfig`` command?

Drkstr, 2021-05-15:
HI @stephendade, apologies for the delayed reply. This is a weekend project am working on. 

Here the ifconfig output with the dongle plugged in and active(status lights blue or green)

`eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:ac:84:cf  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.2.100  netmask 255.255.255.0  broadcast 10.0.2.255
        inet6 fe80::dffa:77fb:bbed:744  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:ac:84:d0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 5228 (5.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0`

For comparison,  here what ifconfig looks like before I installed Rpanion server

`eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:ac:84:cf  txqueuelen 1000  (Ethernet)
        RX packets 155  bytes 52923 (51.6 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 96  bytes 11130 (10.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.100  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::d58:c483:984:20bc  prefixlen 64  scopeid 0x20<link>
        inet6 2001:8004:2748:f07:79a3:4b41:c8e3:7e18  prefixlen 64  scopeid 0x0<global>
        inet6 2001:8004:2748:f07:b827:c5b4:b86b:2  prefixlen 128  scopeid 0x0<global>
        ether 00:1e:10:1f:00:00  txqueuelen 1000  (Ethernet)
        RX packets 3976  bytes 5101191 (4.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2462  bytes 205562 (200.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:ac:84:d0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0`

Let me know if you would like anymore info.

stephendade, 2021-05-16:
Ok. For some reason the dongle isn't being picked up as a network connection.

I haven't got this dongle to test, so there's not much I can do to help. You'll probably have better luck for the issue and seeing if someone else has solved it.

Drkstr, 2021-05-16:
Hi @stephendade, looks like having the dongle plugged when the pi is booting up makes it work :)

closed 2021-05-16 by Drkstr