stephendade/Rpanion-server #28: Sending NTRIP data to PixHawk using Rpanion-server

Drkstr, 2021-05-16:
I have a Raspberry Pi companion computer working with a PixHawk. 

Would it be possible to add an NTRIP component to the Rpanion project? 

Or can we use the Rpanion server's MAVlink connection to send correction signals from an NTRIP (client on the Pi) to the PixHawk

stephendade, 2021-05-17:
> Would it be possible to add an NTRIP component to the Rpanion project?

I'll add it to the to-do list

Drkstr, 2021-07-03:
Hi @stephendade, I'd like to have a crack at this myself. 

Would you please give me some pointers for how I might get started?

davidbuzz, 2021-07-04:
I'd suggest that you need to run the 'str2str' tool from RTKLib  http://www.rtklib.com/  as a basic acommand-line setup.

Or...   you might find that this works for you:

https://github.com/Stefal/rtkbase

stephendade, 2021-07-04:
> Would you please give me some pointers for how I might get started?

Best architecture would be a Python script to grab the NTRIP messages, format to mavlink, and send over UDP to mavlink-router. For the frontend, a new page on the server GUI to start/stop the NTRIP script and edit settings.