stephendade/Rpanion-server #31: System stats

stephendade, 2021-06-12:
Add warning when system temperature, voltage or averaged CPU exceeds limits.

This would allow users to diagnose some stability issues, especially when running video streaming

lorenzing, 2022-01-26:
Great function! I think we can also integrate the reading of undervoltage and cpu throtteling.
```
vcgencmd get_throttled
vcgencmd measure_temp
vcgencmd measure_volts core
vcgencmd measure_clock core
```
The `vcgencmd get_throttled` output bits are:
[image]

Other commands can be integrated, such as reading the gpu or free ram.
`vcgencmd get_mem arm/gpu`

For example this is the dashboard of my Raspberry Pi (Zero 2 W) which controls the 3D printer:
[image]

stephendade, 2022-01-29:
I'd prefer a more generic approach, as Rpanion-server also runs on Jetsons and a few x86 platforms now. Is there a more generic functions we can use?

lorenzing, 2022-02-01:
Hello Stephen, sorry for the delay in replying. You are absolutely right about the compatibility of the feature with all supported platforms. However, I believe it is important to know if the raspberry get throttled.

stephendade, 2022-02-01:
> However, I believe it is important to know if the raspberry get throttled.

Yes, that's fine. I'm generally happy to put in small amounts of Pi-specific functionality

ddd999, 2024-03-07:
lm-sensors is probably the most widely supported way to get hardware monitoring info: https://github.com/lm-sensors/lm-sensors