stephendade/Rpanion-server #34: Feature request: Add possibility to enable/set listening TCP port of mavlink router

nwalser, 2021-07-06:
It would be awesome to be able to enable/set the listening port of the mavlink-router.

This would be required in a scenario where to GCS wants to connect via WLAN network directly to the mavlink network.

GCS -> TCP (over WLAN) -> Rpanion-server

Please let me know if I got something wrong in the architecture and it is not intended to use it this way.
For now I got it working by changing the line with the desired port:

https://github.com/stephendade/Rpanion-server/blob/fcc8e872001a492f8a465d0fa120d3af92296fdf/server/flightController.js#L252

stephendade, 2021-07-06:
Is there a particular reason you want to use TCP? Mavlink telemetry is almost always sent over UDP (outside of simulation)

nwalser, 2021-07-07:
I just wanted to test if the  rpanion-server works correctly. And to do that I expected that I could connect with the rpanion server via TCP without the need to run a router software that the drone connects to on my computer.

In my final setup I wont use this option, because it will look more like this:
Pixhawk 4 (Ardupilot) ---(UART)--> RaspberryPi Zero (Rpanion-server) ---(UDP)--> Server on the internet ---(UDP)--> GCS

I thought it would be usefull in the case when you want to connect to the drone via the WLAN without the need to run any software on your PC (except the GCS of your choice).

stephendade, 2021-07-08:
Ok, I'll look at adding that option.

PR #35 (2021-09-19): Fixes
Fixes #34 and #28, among other things
closed 2021-09-19 by stephendade