stephendade/Rpanion-server #50: node server exit: [nodemon] clean exit - waiting for changes before restart

TyphoonWu, 2021-12-06:
pi3b+ flash Rpanion-Server-0.7.0-RasPiOS.img.xz. 
The '/usr/bin/node /home/pi/Rpanion-server/server' work well.
Normal processes list:
pi@rpanion:~ $ ps -Af|grep node
pi         438   390  0 17:17 ?        00:00:00 sh -c node-env-run server --exec nodemon | pino-colada
pi         439   438  0 17:17 ?        00:00:02 node /home/pi/Rpanion-server/node_modules/.bin/node-env-run server --exec nodemon
pi         440   438  0 17:17 ?        00:00:01 node /home/pi/Rpanion-server/node_modules/.bin/pino-colada
pi         541   439  3 17:17 ?        00:00:09 node /home/pi/Rpanion-server/node_modules/.bin/nodemon /home/pi/Rpanion-server/server
pi         554   541  2 17:17 ?        00:00:08 /usr/bin/node /home/pi/Rpanion-server/server
pi         858   825  0 17:22 pts/2    00:00:00 grep --color=auto node

But node server cannot work after the steps:
1. cd Rpanion-server
2. git pull (update master branch)
3. ./deploy/RasPi2-3-4-deploy.sh

Logs:
-- Logs begin at Sun 2021-12-05 16:46:56 GMT, end at Mon 2021-12-06 14:28:46 GMT. --
Dec 05 16:47:00 rpanion systemd[1]: Started Rpanion Server Service.
Dec 05 16:47:04 rpanion npm[399]: > Rpanion-server@0.7.0 server /home/pi/Rpanion-server
Dec 05 16:47:04 rpanion npm[399]: > node-env-run server --exec nodemon | pino-colada
Dec 05 16:47:07 rpanion npm[399]: [nodemon] 2.0.15
Dec 05 16:47:07 rpanion npm[399]: [nodemon] to restart at any time, enter `rs`
Dec 05 16:47:07 rpanion npm[399]: [nodemon] watching path(s): *.*
Dec 05 16:47:07 rpanion npm[399]: [nodemon] watching extensions: js,mjs,json
Dec 05 16:47:07 rpanion npm[399]: [nodemon] starting `node /home/pi/Rpanion-server/server`
Dec 05 16:47:10 rpanion npm[399]: [nodemon] **clean exit - waiting for changes before restart**

//failure processes list:
pi@rpanion:~/Rpanion-server $ ps -Af|grep node
pi         426   399  0 14:26 ?        00:00:00 sh -c node-env-run server --exec nodemon | pino-colada
pi         427   426  0 14:26 ?        00:00:01 node /home/pi/Rpanion-server/node_modules/.bin/node-env-run server --exec nodemon
pi         428   426  0 14:26 ?        00:00:00 node /home/pi/Rpanion-server/node_modules/.bin/pino-colada
pi         451   427  0 14:26 ?        00:00:04 node /home/pi/Rpanion-server/node_modules/.bin/nodemon /home/pi/Rpanion-server/server
pi         811   769  0 14:33 pts/1    00:00:00 grep --color=auto node

stephendade, 2021-12-06:
Can you post the last 20 lines of ``~/Rpanion-server/logs/app.log``?

TyphoonWu, 2021-12-08:
C

closed 2021-12-08 by TyphoonWu