stephendade/Rpanion-server #54: Video streaming settings

scala93, 2021-12-22:
I've just tested out the new release. A lot of improvements! 
But for the streaming section both resolution and UDP IP address reset to default at every start up. I tried to set my own values and reboot as the 0.8 guide says but it did not work.

stephendade, 2021-12-23:
Thanks for reporting this - It's now been fixed.

lorenzing, 2021-12-24:
Hi, are the ISOs already updated?

stephendade, 2021-12-25:
> Hi, are the ISOs already updated?

The ISOs are only updated for major releases. Follow the instructions at https://github.com/stephendade/Rpanion-server#updating to update to the latest code of Github

scala93, 2021-12-26:
Stephen I've also found out that when changing resolution it zoomes instead of really changing resolution.
Moreover, once I disconnect the connection it does not restart the video link (using zerotier and qgc).

stephendade, 2021-12-26:
> Stephen I've also found out that when changing resolution it zoomes instead of really changing resolution.

What camera is this with? The Raspberry Pi Camera does in some cases crop/zoom the image, depending on the selected resolution and framerate (see https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes)

scala93, 2021-12-26:
> What camera is this with? The Raspberry Pi Camera does in some cases crop/zoom the image, depending on the selected resolution and framerate (see [...]

It is a raspberry pi camera V2. And it's cropping as resolution goes higher. So no solution for that? And for the non reconnection issue what do you think?

stephendade, 2021-12-29:
> And for the non reconnection issue what do you think?

That should now be fixed. Give it a test and see if that works for you.

> It is a raspberry pi camera V2. And it's cropping as resolution goes higher. So no solution for that?

That's a function of the camera itself. Nothing I can do to fix that.

scala93, 2021-12-29:
> That should now be fixed. Give it a test and see if that works for you.
>
>
> That's a function of the camera itself. Nothing I can do to fix that.

From bench test streaming video reconnection issue has been solved.

closed 2022-01-22 by scala93