stephendade/Rpanion-server #59: [Feature Request] VPN web interface

lorenzing, 2021-12-29:
Hello, 
hoping to propose something useful for other users, I wanted to ask if you can insert a VPN page in the web interface. 
As a vpn client I use vpnc which to work takes data from /etc/vpnc/rpanion.conf within which they must be specified:
```
IPSec gateway myhost.net
IPSec ID vpn_user
IPSec secret TANrpS1y34hHHJGIS4
IKE Authmode psk
Xauth username vpn_user
Xauth password <passwort>
local port 0
DPD idle timeout (our side) 0
```
The tunnel test command are:
`sudo vpnc rpanion.conf
`
To stopping tunnel:
`sudo vpnc-disconnect
`

If you could edit the values of this file in the web interface it would be very convenient for me without having to connect with ssh.
I have also seen that several people use Zero-Tier and you could also enter for them in this page the possibility to put their NETWORKID.

I insert a sample to explain myself better, taking inspiration from a service that already exists:
[image]

Thank you so much anyway, Lorenzo.

giyowiy, 2022-01-22:
Wow that would be a great feature! Hope it can be implemented soon

Gyoway

stephendade, 2022-01-23:
Sure - I'll look at adding that in the next few months.

lorenzing, 2022-01-23:
Thank you so much @stephendade and congratulations on your great work!

hyperion11, 2022-09-27:
hello! when version with VPN will be released?

stephendade, 2022-09-28:
I'm actually writing the VPN feature at the moment, for Wireguard and Zerotier. I'm hoping it'll be done in the next month.

closed 2022-11-19 by lorenzing