stephendade/Rpanion-server #69: V 0.9 to do list

stephendade, 2022-02-09:
This list will track the to-be-completed features/fixes for V0.9:

- [ ] Add VPN interface
- [ ] Cloud upload of logs and other files
- [ ] Multi-camera streaming
- [ ] Option to add username/password login to Rpanion

stephendade, 2022-09-28:
Given it's been a long time since the last release and I'm finding I have less time these days, I'll cut the feature list for V0.9 to what is currently done, plus a VPN GUI for Wireguard and Zerotier.

Plan is to complete the VPN GUI in the next couple of weeks, then start beta testing V0.9

The uncompleted features in the list I'll defer to a future release.

stephendade, 2022-11-11:
Ok, VPN GUI is done :)

That's all major features done for V0.9, so I'll start with detailed checks and testing on a Pi (Ras Pi OS and latest LTS Ubuntu) and (possibly) a Jetson Nano.

mohammada315, 2022-11-19:
We are eager to use as soon as possible
thank you Rpanion developers

stephendade, 2022-12-18:
0.9 is released, so closing. Will make a new list for V0.95

closed 2022-12-18 by stephendade