stephendade/Rpanion-server #80: Rpanion Login

ProcessorGuy, 2022-04-19:
Hi,

After installing and working properly ardupilot software, I installed the rpanion server through ssh. Then, the system was rebooted. After, I couldn't reach the raspberry pi again. 

Then, I connected to a micro HDMI cable and monitor to see what was going on there. The system requested me to enter login credentials. There was a statement that said ` The hotspot has the SSID "rpanion" and password "rpanion123". ` However, I believe this is for wifi. What are the login credentials?  

https://ibb.co/ftz1Xb8

Also, I couldn't use my keyboard to type the credentials, normally I could use it. Thanks.

stephendade, 2022-04-20:
The SSH credentials are unchanged from the system defaults. For Raspian, this is usually "pi" and "raspberry"

ProcessorGuy, 2022-04-20:
Hi again. I couldn't reach the raspberry pi through ssh. I checked the .sh files, and I saw this line ` sudo apt purge -y openresolv dhcpcd5 modemmanager ` . I believe this causes a problem. When we purge dhcpcd5, it doesn't assign any IP to my raspberry pi, and I cannot reach raspberry pi through ssh. 

Does it (dhcpcd5) has to be removed or purged in order to run this rpanion server? 

Also, I reinstalled the emild-navio2 image from their website without building rpanion server. It automatically assigned my IP address so that I could easily reach the raspberry pi through ssh.

https://ibb.co/BrRCQCG

stephendade, 2022-04-21:
The networking management features in Rpanion-server use ``nmcli``, which conflicts with ``dhcpcd5``, thus the need to remove ``dhcpcd5``.

If you don't want to use the networking features in Rpanion-server, you can install ``dhcpcd5``.

I've not worked with the emild-navio2 image, so I can't offer specific support for that setup.

ProcessorGuy, 2022-04-21:
Before the reboot which is the last statement of the RasPi2-3-4-deploy.sh, I am planning to add the following commands;

sudo nmcli con mod enps03 ipv4.addresses 192.168.0.101/24
sudo nmcli con mod enps03 ipv4.gateway 192.168.0.1
sudo nmcli con mod enps03 ipv4.dns “8.8.8.8”
sudo nmcli con mod enps03 ipv4.method manual
sudo nmcli con up enps03

note: enps03 (for ethernet connection, not wifi) is the network name of the ethernet, 192.168.0.101 is my desired static IP, “8.8.8.8” is the desired DNS server, the manual indicates that our IP is static, and up means apply our changes now.

I will apply and let you know if it works. Thanks for your help.

ProcessorGuy, 2022-04-22:
It worked but I had to change `/etc/NetworkManager/NetworkManager.conf` file as follows,

[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=true

If I didn't change the file like above, wlan0 would always be unavailable. Also, I did one minor change as follows,

`sudo mv /usr/lib/NetworkManager/conf.d/10-globally-managed-devices.conf  /usr/lib/NetworkManager/conf.d/10-globally-managed-devices.conf_orig`

`sudo touch /usr/lib/NetworkManager/conf.d/10-globally-managed-devices.conf`

Then, I typed as follows,

`sudo systemctl restart NetworkManager`

I was able to access rpanion server. Thank you for your help. Have a nice day !

closed 2022-05-15 by stephendade