stephendade/Rpanion-server #82: Web interface not working

scala93, 2022-05-07:
I have done an update and web interface is not working on http://10.0.2.100:3000/

stephendade, 2022-05-09:
Can you do a ``systemctl status rpanion.service --no-pager --full`` and post the output? It'll help me narrow down what the issue is.

scala93, 2022-05-10:
● rpanion.service - Rpanion Server Service
   Loaded: loaded (/etc/systemd/system/rpanion.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-12-19 04:42:25 GMT; 4 months 20 days ago
 Main PID: 421 (npm)
    Tasks: 37 (limit: 718)
   CGroup: /system.slice/rpanion.service
           ├─421 npm
           ├─588 sh -c node-env-run server --exec nodemon | pino-colada
           ├─589 node /home/pi/Rpanion-server/node_modules/.bin/node-env-run server --exec nodemon
           ├─590 node /home/pi/Rpanion-server/node_modules/.bin/pino-colada
           └─603 node /home/pi/Rpanion-server/node_modules/.bin/nodemon /home/pi/Rpanion-server/server

May 10 08:25:28 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:28 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:28 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:28 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:28 rpanion npm[421]: [nodemon] starting `node /home/pi/Rpanion-server/server`
May 10 08:25:29 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:29 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:29 rpanion npm[421]: [nodemon] restarting due to changes...
May 10 08:25:30 rpanion npm[421]: [nodemon] starting `node /home/pi/Rpanion-server/server`
May 10 08:25:34 rpanion npm[421]: [nodemon] clean exit - waiting for changes before restart

stephendade, 2022-05-15:
Thanks for that - I'll develop a fix in the next week or two.

closed 2022-05-18 by stephendade

stephendade, 2022-05-18:
I've tested this and it show now work. You'll need to manually run ``deploy/upgrade.sh`` in the Rpanion-server folder to re-build and restart the server.

scala93, 2022-05-19:
I confirm now web interface is working. I can see new features working as well.

Saluti,

Francesco Scalabrin

PR #86 (2022-05-18): Better upgrade UI
Fixes #82