stephendade/Rpanion-server #98: The possibility of forward Mavlink in serial port

mohammada315, 2022-08-10:
dear @stephendade 
Is it possible to send data to multiple telemetry?

stephendade, 2022-08-10:
I'm not too sure what you're referring to. Do you want to send telemetry Flight Controller -> Serial Port -> Companion Computer -> Different Serial Port -> Some Device ?

At this time, telemetry can only be sent Flight Controller -> Serial Port -> Companion Computer -> Network (TCP/UDP)

mohammada315, 2022-08-11:
i want this one

Flight Controller -> USB -> Companion Computer

Companion Computer -> different telemetry -> multiple device -> ground station

https://s6.uupload.ir/files/diagram_d97.png

stephendade, 2022-08-11:
Ok, I understand.

No, not currently supported. Rpanion-server can only forward to TCP/UDP at this time.

It's possible to add support (as the mavlink-router software does support it), but not something I'm planning at this time.

mohammada315, 2022-08-12:
Thanks a lot @stephendade

closed 2022-08-12 by mohammada315

bajwasarim, 2023-07-22:
@mohammada315 

> i want this one
> 
> Flight Controller -> USB -> Companion Computer
> 
> Companion Computer -> different telemetry -> multiple device -> ground station
> 
> https://s6.uupload.ir/files/diagram_d97.png

You can use MavProxy for this purpose. Add an extra UDP out address on localhost:UDPport in mavrouter-d service run by Rpanion server. Use this udp port as an input to MavProxy and forward the MAVlink traffic to the companion computer