stephendade/Rpanion-server #100: .bin log file missing information

sonph85058, 2022-09-14:
Hi,
I'm using the Flight Logs feature to record the Bin logs from the ardupilot AP (H743 mini board, serial 2 port connect to a Pi Zero 2, baudrate 921000). I've changed the LOG_BACKEND_TYPE to 3 (File and MAVLink), LOG_BITMASK was 65535 and got the log files in the Pi. However, the log files seem to have only full data of information that I had displayed on the ground station screen (in this case are the range finder and battery info). Other information such as raw GPS data, IMU,.. etc were missing or having very little information (4-5 data point only). Here is a log sample: https://1drv.ms/u/s!AuE6w7rDGjlugtxiouMWLMdMPMxSFA?e=HEzPwS
[image]

Is there any step in setting I am missing to be able to have the full information data log? 
Thanks in advance!

stephendade, 2022-09-14:
It's possible that the logging data is too much for the serial link, and some packets are being missed. Try removing some items from the ``LOG_BITMASK`` and trying again.

Or you could try a faster baud rate.

I'm travelling at the moment, but I'll take a look next week if you've not already solved it by then.

stephendade, 2023-04-06:
Covered in #122

closed 2023-04-06 by stephendade