stephendade/Rpanion-server #103: [Feature Request] add option to stream video from IP camera

hyperion11, 2022-10-26:
Good afternoon. It would be interesting to add the ability to stream from the IP camera. 
Like ZR10 http://en.siyi.biz/en/Gimbal%20Camera/
I try simple IP camera with zerotier and get it work. But probably not as beautiful as we would like
What i do.
1. Connect IP camera over Ethernet cable to ETH port in RPI. IP address camera 192.168.144.25, RPI ETH 192.168.144.1
2. Install Zerotier VPN on both sides and add route to 192.168.144.0/24 subnet via Zerotier's address RPI
3. sudo sysctl -w net.ipv4.ip_forward=1 on RPI
4. now i have access to 192.168.144.25 from Zerotier tunnel

Maybe there is a more elegant solution?

timtuxworth, 2022-10-26:
> Good afternoon. It would be interesting to add the ability to stream from the IP camera. Like ZR10 http://en.siyi.biz/en/Gimbal%20Camera/ I try simple IP camera which i have at this moment (https://www.xiongmaitech.com/en/index.php/product/product-detail/185/187/413) with zerotier and get it work. But probably not as beautiful as we would like What i do.
> 
> 1. Connect IP camera over Ethernet cable to ETH port in RPI. IP address camera 192.168.144.25, RPI ETH 192.168.144.1
> 2. Install Zerotier VPN on both sides and add route to 192.168.144.0/24 subnet via Zerotier's address RPI
> 3. sudo sysctl -w net.ipv4.ip_forward=1 on RPI
> 4. now i have access to 192.168.144.25 from Zerotier tunnel

This is _very_ interesting. Can you provide the specific command you used at step 2?

hyperion11, 2022-10-27:
[image]
i add route here. And it apply automatically when establishing vpn connection

makeflyeasy, 2025-09-26:
We are looking forward to adding the function of IP network cameras as video output sources so that more cameras can be used.

stephendade, 2025-10-25:
This feature has now been added