stephendade/Rpanion-server #111: Camera not recognized Arducam IMX477

mohammada315, 2022-11-26:
dear @stephendade 
On the video streaming page In the device field
I only have the test source

Is there a way to add arducam imx477 camera?

stephendade, 2022-11-27:
Only the IMX219 (Official Pi Cam V2) is supported via CSI. Most USB cameras are supported.

mohammada315, 2022-11-27:
Is there any way to use IMX477?

mohammada315, 2022-11-27:
I tried on Jetson NX and Raspberry V4
But the camera was not found

stephendade, 2022-11-27:
> Is there any way to use IMX477?
> 
> on my device using the pipline libcamerasrc It works alone
> 
> 
> 
> 

That's a good start. I've actually had to disable libcamerasrc cameras for now, as I've had significant stability issues with the the driver. I plan to add it back once I've figured out the issues.

mohammada315, 2022-12-03:
I ran it as a commandline, there was no problem
thank you

closed 2022-12-03 by mohammada315