stephendade/Rpanion-server #119: Temporary failure resolving 'ports.ubuntu.com'

danylevskyi, 2023-01-22:
**Raspberry Pi 4 Model B**
**Ubuntu Server 20.04**

```sh
Err:1 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe armhf wireguard-tools armhf 1.0.20200513-1~20.04.2
  Temporary failure resolving 'ports.ubuntu.com'
Err:2 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe armhf wireguard all 1.0.20200513-1~20.04.2
  Temporary failure resolving 'ports.ubuntu.com'
Err:3 http://ports.ubuntu.com/ubuntu-ports focal/universe armhf resolvconf all 1.82
  Temporary failure resolving 'ports.ubuntu.com'
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/w/wireguard/wireguard-tools_1.0.20200513-1~20.04.2_armhf.deb  Temporary failure resolving 'ports.ubuntu.com'
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/w/wireguard/wireguard_1.0.20200513-1~20.04.2_all.deb  Temporary failure resolving 'ports.ubuntu.com'
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/r/resolvconf/resolvconf_1.82_all.deb  Temporary failure resolving 'ports.ubuntu.com'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

stephendade, 2023-01-29:
That error message occurs if there are network connection issues to the Ubuntu apt repository.

You'll need to run the script again

stephendade, 2023-03-27:
Fixed in #132

closed 2023-03-27 by stephendade