stephendade/Rpanion-server #121: Deploy fails when making gst-rpicamsrc

TECH7Fox, 2023-01-31:
When doing `./RasPi2-3-4-deploy.sh` it fails when making gst-rpicamsrc.

```
make  all-recursive
make[1]: Entering directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc'
Making all in src
make[2]: Entering directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc/src'
make  all-am
make[3]: Entering directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc/src'
  CC       libgstrpicamsrc_la-gstrpicam-enum-types.lo
In file included from gstrpicam-enum-types.c:6:
gstrpicam_types.h:1:10: fatal error: interface/mmal/util/mmal_util_params.h: No such file or directory
    1 | #include "interface/mmal/util/mmal_util_params.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Makefile:492: libgstrpicamsrc_la-gstrpicam-enum-types.lo] Error 1
make[3]: Leaving directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc/src'
make[2]: *** [Makefile:375: all] Error 2
make[2]: Leaving directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc/src'
make[1]: *** [Makefile:408: all-recursive] Error 1
make[1]: Leaving directory '/home/admin/Rpanion-server/modules/gst-rpicamsrc'
make: *** [Makefile:340: all] Error 2
```

My OS:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
```

Would be great if there is a fix for this.
I also saw that gst-rpicamsrc is now build into GStreamer from 1.18. Could this be used perhaps? Willing to make a PR if you point me in the right direction.

stephendade, 2023-01-31:
> I also saw that gst-rpicamsrc is now build into GStreamer from 1.18. Could this be used perhaps?

Yes, that would be the preferred solution. I _think_ that only Bullseye has Gstreamer >1.18, so anyone on an older OS will need to build gst-rpicamsrc from source.

> Willing to make a PR if you point me in the right direction

Sure - that'd be great! There'd need to be a check in the deploy scripts to only build gst-rpicamsrc if the gstreamer version is too old. In the longer term I'll likely remove the gst-rpicamsrc module and users will be required to use Bullseye or later.

As for your original error - are you running the 64-bit version of RasPiOS? I don't think gst-rpicamsrc builds correctly on 64-bit systems. Otherwise, check that ``/usr/include/interface/mmal/util`` exists and contains the relevant header files.

TECH7Fox, 2023-02-05:
Thanks for your answer. Will see what I can do, but currently I'm having some other issues with my camera module.

Yes I'm using the 64-bit lite version. But with the other fix it's fine.

stephendade, 2023-03-22:
For reference, I've now removed the build for ``gst-rpicamsrc`` and updated the documentation to tell users to either a) Use libcamera b) use the latest Bullseye with Gstreamer >1.18 using the legacy interface or c) build ``gst-rpicamsrc`` themselves

closed 2023-04-28 by stephendade