stephendade/Rpanion-server #122: .bin logger bug

MorgothT, 2023-02-21:
I have a boat with Mosaic-H GPS (outputs gga, rmc, hdt at 10Hz, according to simpleRTK instructions)
when testing indoors the logging looks OK, but when outside and the gps get a fix the .bin logging restarts every 6Kb.
a new file is started and the older ones are gone.

is this a baud rate issue ?
i will try next time to log only the rangefinder and gps.

stephendade, 2023-02-21:
That's a known issue in the backend mavlink-router, see https://github.com/mavlink-router/mavlink-router/pull/318

The 6Kb files are blank logfiles, which Rpanion-server will clean up.

If you're just getting the 6Kb files, then ArduPilot is not streaming the binlog messages to Rpanion-server. It might be a baud issue is you running less than 921600 baud.

MorgothT, 2023-02-22:
since the baud rate is 921600 and it's logging fine when the GPS has no sats, I assume data is truncated at that baud rate.
I'll try to remove unnecessary data streams and increase the baud rate.

thank you for the quick answer.

MorgothT, 2023-03-20:
I've changed the baud rate to 1500k but the problem persists.
Tried messing with the mavlink buffer and when the buffer was lower then 8k the logs were at different size (8K,10K) but still "empty", when the buffer was 16k it didn't log at all.
I've noticed that the FC has low memory free about 20k (it's cube black), could this be a problem ?
When i connect the pi using usb it works ok (even at 115k).
Btw, even with the log mask only for rangefinder and gps I still get a lot of XK logs.

stephendade, 2023-03-22:
Maybe try 115k on the uart.

I know my KakuteF4 can only to up to 256k baud with basic logging. Anything more and it can't keep up.

Are you getting any packet loss on the link?

MorgothT, 2023-04-02:
Looks like that fixed the issue I was having.
Thank you.

closed 2023-04-02 by MorgothT