stephendade/Rpanion-server #125: Is the TLOG generation broken?

brunoolivieri, 2023-03-14:
Hello :-)

I have been having problems with the TLOGS. I think TLOGS are being generated without GPS coordinates inside Rpanion.

Initially, it seemed like an intermittent thing. At the moment, it is more common to have no GPS data on them than to have the data correctly entered. This problem causes the KMZ to be generated broken.

The BIN logs taken directly from the flight controller show the flights performed correctly.

I made a set of repeat flights and joined the specific binlogs, tlogs and kmz: https://drive.google.com/file/d/14MBc0TJDwMZ1bEXylL6Cyf7xclrfh0Lf/view?usp=share_link

Can you please help me?

stephendade, 2023-03-22:
Yes, there an issue with missing messages in the tlogs. It's related to the mavlink parser not parsing and logging all packets. I'll hopefully have a fix soon.

brunoolivieri, 2023-03-22:
Great! I will be looking for it. Thank you!

stephendade, 2023-03-29:
Can you test #131? That should fix it

brunoolivieri, 2023-03-30:
Sure!

brunoolivieri, 2023-04-05:
Hi!

I did some tests and none of them resulted on TLOGs with GPS points inside: (so, the KMZs are empty as well)

[image]

Could you check it out?

stephendade, 2023-04-06:
The Log bitmask is only for the .bin logfiles, not the tlogs.

Are you running with a GCS connected? There won't be any telemetry data is there's no GCS connected.

brunoolivieri, 2023-04-11:
Do you mean no GPS data in the TLOGs without a GCS?   

These pieces of information were inside the TLOGs when the KMZ logs were tested. I guess I still have some of them working here: http://201.49.23.53:3000/flightlogs

Regarding the "node-js-mavlink" connection showing messages on the /controller inside the "console output": Does it count as a GCS, am I right?

stephendade, 2023-04-12:
Are you connecting to a ground station during your flights? Rpanion-server relies on the ground station to send the ``set_message_interval`` MAVLink messages that ArduPilot needs to start full tlog streaming.

I should _probably_ add an option for Rpanion-server to start the streaming itself...

brunoolivieri, 2023-04-12:
I could connect some ground stations during old tests, but not on these flights with the link above. On real flights, I couldn't use it.

Regarding the Rpanion-server starting setting the set_message_interval: Why should it be an option? Is there any issue with always starting it? It is the first connection with the ArduPilot; any ground station would set it later. Is that a risk?

stephendade, 2023-04-13:
> Regarding the Rpanion-server starting setting the set_message_interval: Why should it be an option? Is there any issue with always starting it? It is the first connection with the ArduPilot; any ground station would set it later. Is that a risk?

Only issue is if the user need a slower message rate for a particular reason. The initial (higher) rate set by Rpanion-server might disrupt things.

I'll add an option for Rpanion-server to set the message rate.

stephendade, 2023-04-14:
@brunoolivieri: I've added a datastream request option to #131. It's a checkbox in the flight controller page.

See if that works for you.

brunoolivieri, 2023-04-16:
Tks, I will test and let you know for sure.

brunoolivieri, 2023-04-25:
I spent a couple of batteries, and #131 worked well! THANKS!

closed 2023-04-25 by stephendade