stephendade/Rpanion-server #128: bluetooth needs to be disabled on Pi3 for the UART to work properly

altmattr, 2023-03-17:
I added
> dtoverlay=pi3-miniuart-bt

to `boot/config.txt` and rebooted to make it work.

It's needed because the Pi3 will run some bluetooth stuff on the same serial interface as the telemetry.

stephendade, 2023-03-18:
Is there any reason you can't enable the uart via the ``enable_uart=1`` dtoverlay?

altmattr, 2023-03-18:
I was using a pre-built image and it just wasn't working.  The same image had worked perfectly on a Pi2 but that had no wifi so I had to upgrade to the Pi3 I found lying around.  I was getting some packets through, but it was not working right.  Then I remembered that on the pi3 the bluetooth uses the same serial so you get junk in the stream.  I added the extra overlay and suddenly everything was perfect.  I have to put in this overlay on every Pi3 when I try to use a UART terminal on GPIO or I get junk appearing on that terminal as well.

altmattr, 2023-03-18:
but the subtleties of what each boot config.txt does is beyond me

stephendade, 2023-03-18:
Is that a Pi 3B or 3B+?

I've got a 3B+ here, and the ``enable_uart=1`` works fine for assigning ``/dev/serial0`` to GPIO14/15 on the Pi

altmattr, 2023-03-18:
I'll try that myself then next time I get the chance, which will be sometime next week.  Do you think that boot option might do more than the equivalent `raspi-config` line (which is already in your deployment script)?

stephendade, 2023-03-18:
>  Do you think that boot option might do more than the equivalent raspi-config line (which is already in your deployment script)?

The ``sudo raspi-config nonint do_serial 2`` does indeed add the ``enable_uart=1`` to ``/boot/config.txt``, plus disables the console on that port.

altmattr, 2023-03-18:
this page https://www.circuits.dk/setup-raspberry-pi-3-gpio-uart/ seems to explain the behaviour I see.

altmattr, 2023-03-22:
I've done a test from scratch and I can confirm that on my pi3, using a pi connect lite the extra overlay is required

Without it you get a low byte rate and no packets. With it you get working telemetry. 

For this test I used a clean pios lite, cloned the repo, used the scripts to setup and then connected to the flight controller

stephendade, 2023-03-22:
Thanks for looking into that.

I've just tested here with a Pi 3B, Rpanion-Server 0.9.0 disk image and Pi-Connect Lite. No telemetry issues at 115200 baud.

Are you attaching any additional devices to the Pi?

stephendade, 2023-04-06:
I've added a note to the documentation (https://www.docs.rpanion.com/software/rpanion_server_v09#flight_controller), in case other users come across this issue.

closed 2023-04-06 by stephendade