stephendade/Rpanion-server #130: Installation failed

andyp1per, 2023-03-20:
Running ```./RasPi2-3-4-deploy.sh``` it got quite far and then ended with:

```
The following NEW packages will be installed:
  libc-ares2 libjs-highlight.js libnode72 nodejs nodejs-doc
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 10.9 MB of archives.
After this operation, 47.9 MB of additional disk space will be used.
Ign:1 http://deb.debian.org/debian bullseye/main arm64 libc-ares2 arm64 1.17.1-1+deb11u1
Err:2 http://security.debian.org/debian-security bullseye-security/main arm64 libnode72 arm64 12.22.12~dfsg-1~deb11u3
  Temporary failure resolving 'security.debian.org'
Ign:3 http://deb.debian.org/debian bullseye/main arm64 libjs-highlight.js all 9.18.5+dfsg1-1
Err:4 http://security.debian.org/debian-security bullseye-security/main arm64 nodejs arm64 12.22.12~dfsg-1~deb11u3
  Temporary failure resolving 'security.debian.org'
Ign:5 http://security.debian.org/debian-security bullseye-security/main arm64 nodejs-doc all 12.22.12~dfsg-1~deb11u3
Err:3 http://deb.debian.org/debian bullseye/main arm64 libjs-highlight.js all 9.18.5+dfsg1-1
  Temporary failure resolving 'deb.debian.org'
Err:1 http://deb.debian.org/debian bullseye/main arm64 libc-ares2 arm64 1.17.1-1+deb11u1
  Temporary failure resolving 'deb.debian.org'
Err:5 http://security.debian.org/debian-security bullseye-security/main arm64 nodejs-doc all 12.22.12~dfsg-1~deb11u3
  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/c/c-ares/libc-ares2_1.17.1-1%2bdeb11u1_arm64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/h/highlight.js/libjs-highlight.js_9.18.5%2bdfsg1-1_all.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/n/nodejs/libnode72_12.22.12%7edfsg-1%7edeb11u3_arm64.deb  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/n/nodejs/nodejs_12.22.12%7edfsg-1%7edeb11u3_arm64.deb  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/n/nodejs/nodejs-doc_12.22.12%7edfsg-1%7edeb11u3_all.deb  Temporary failure resolving 'security.debian.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

stephendade, 2023-03-20:
Looks like a DNS failure. Maybe check your network's DNS settings?

andyp1per, 2023-03-21:
I am able to ping these names ok - can I restart the install process?

stephendade, 2023-03-21:
Yes - just run the script again

stephendade, 2023-03-22:
This _may_ be an issue with the deploy script, as there's some networking config going in the background. I do sometimes get the same errors as you did.

I'll see if I can isolate the issue

PR #132 (2023-03-27): Deploy: Install openresolv last
Turns out there _was_ an issue with the deploy scripts. The ``openresolv`` package messes up the DNS configuration, failing subsequent deploy steps.

Fixes #130 and #119
closed 2023-03-27 by stephendade