stephendade/Rpanion-server #134: HOW to connect WWW   by   wifi?

zouzhe1, 2023-04-12:
I see that your document only has WiFi hotspots, 
but WiFi hotspots allow other devices to connect to you.
 I want to send the video stream to my server, so I definitely want Raspberry Pi to connect to www. 
Your document did not introduce it.

stephendade, 2023-04-12:
You can disable the hotspot and connect to an existing Wifi network via the "Network" page in Rpanion-Server. See https://www.docs.rpanion.com/software/rpanion_server_v09#network

zouzhe1, 2023-04-13:
@stephendade 
Thank you for your answer，but in "Network" page，i cant find  how  to disable the hotspot。It's not mentioned in the document either。i cant konw the "Wifi Access Point" means，can fill router wifi name  and password ？I think the current meaning is that people can only directly connect to raspberry ，Instead of connecting the Raspberry Pi to the router

zouzhe1, 2023-04-13:
@stephendade 
Thank you for your help. I have successfully connected to the router. Are you in Australia? I'm still wondering why you replied to me so quickly. It turns out that the time difference between the Australian capital and China is only 2 hours. My friend is also going to Australia next month, haha.

I'm sorry to ask you another question, how can video streams be pushed to my server? My server has already installed the RTSP forwarding program. Just push the RTSP video stream to the specified IP and port by Raspberry Pi.

stephendade, 2023-04-13:
> I'm sorry to ask you another question, how can video streams be pushed to my server? My server has already installed the RTSP forwarding program. Just push the RTSP video stream to the specified IP and port by Raspberry Pi.

Just connect to the specified RTSP addresses in the ``Video`` tab when streaming is enabled.

zouzhe1, 2023-04-13:
@stephendade 
This Destination IP and Destination port are grayed out and cannot be edited or modified。
If it can be modified here, I think it would be correct to write the IP and port of my server here.
[image]

stephendade, 2023-04-13:
> This Destination IP and Destination port are grayed out and cannot be edited or modified。
If it can be modified here, I think it would be correct to write the IP and port of my server here.

The destination IP and Port are only applicable for RTP streaming mode.

stephendade, 2023-04-28:
#135 hides the IP/port for RTP mode, making it more clear for users

closed 2023-04-28 by stephendade