stephendade/Rpanion-server #148: [Info] Raspberry pi alternative -- Rpanion running on Libre Computer Le Potato

tmarkson, 2023-08-08:
Since Raspberry Pi are expensive and hard to obtain,  I can mention that Rpanion is running well on Le Potato, 2 GB RAM version. 

Rpanion was an easy setup -- thanks for the effort Steven! This one is using Wireguard via 4G USB modem. 

It's about $40 via Amazon Prime here in USA.  The product page is here: 
  https://libre.computer/products/aml-s905x-cc/

This one is using Bullseye for the s905 chip, you can get the OS image.  
Initial setup with a monitor+keyboard.  [Setup info here](https://hub.libre.computer/t/raspbian-11-bullseye-for-libre-computer-boards/82#:~:text=the%20issues%20discovered.-,Headless%20Lite%20Setup,-The%20latest%20Raspbian)
  https://distro.libre.computer/ci/raspbian/11/2023-05-03-raspbian-bullseye-arm64%2Baml-s905x-cc.img.xz

CPU average 0.5 on a quad core while streaming UDP and TCP. 

Funny enough, a Raspi 4 and Le Potato **weigh less** than a Pi zero 2 with power hat and USB OTG cable. 

Note: the Rpanion user was not allowed to access the serial device.  I see the npm service user is `root`, which owns the serial device.  A quick fix was to add a `udev` rule for that serial device to `chmod 666` the serial device. Now the permission is updated after boot and Rpanion automatically starts serving telemetry.  Really great! 

[image]

stephendade, 2023-08-09:
Great! Good to have more boards compatible.

Could you do a PR to:
- Mention the board compatibility in README.md
- Add a script in the deploy folder to add the udev rules

tmarkson, 2023-08-11:
Updated readme in #149. 
Don't want to break something with a new script so I added a link to site with instructions

closed 2023-08-11 by tmarkson