stephendade/Rpanion-server #158: “Adhoc Wifi Config” issues

Ultramarine1939-syujie, 2023-09-04:
I'm testing this program in my laptop, and when I tryied to start a AP hotpot, some error occurs:
1."Error: Error for wireless request "Set Encode" (8B2A) : SET failed on device wlp3s0 ; Invalid argument."
when I set "security" to web and set a password,the error will appear and the hotpot con't work.
2.I can't connect the hotpot when it's "security" is none
I tried to connect with my own phone,but I was refused to connect .

Ultramarine1939-syujie, 2023-09-04:
BTW,my network is Intel AX200.

stephendade, 2023-09-05:
Have you tried the PR in #140? It's got some improvements.

Note, I've found that not all Wifi cards support ad-hoc mode properly under Linux.