stephendade/Rpanion-server #159: [Feature Request] add the option to use Frp for remote connections

Ultramarine1939-syujie, 2023-09-04:
Good morning, I hope there are some page to configure frp server or client for my device.
1.Compared to Zerotier, FRP is simpler and easier to use. It provides intuitive configuration options and documentation, making it very easy to set up and manage FRP. For those unfamiliar with network technology, FRP may be easier to get started.

2.When using FRP, I can fully control the settings and control of internal network penetration. Build your own FRP server without being restricted by third-party services. This can better protect data privacy and security, and have higher controllability for the entire system.

Although I can implement the settings of FRPC through the command line, I think it would be simpler and more practical to add it directly to the script

stephendade, 2023-09-05:
What is FRP?

Ultramarine1939-syujie, 2023-09-06:
you can check it at here:
https://github.com/fatedier/frp