stephendade/Rpanion-server #160: Node.js installation method is deprecated

ddd999, 2023-09-13:
I got this warning when running the RasPi2-3-4-deploy.sh script:

>                           SCRIPT DEPRECATION WARNING
> 
>  This script, located at https://deb.nodesource.com/setup_X, used to
>   install Node.js is deprecated now and will eventually be made inactive.
> 
>   Please visit the NodeSource distributions Github and follow the
>   instructions to migrate your repo.
>   https://github.com/nodesource/distributions
> 
>   The NodeSource Node.js Linux distributions GitHub repository contains
>   information about which versions of Node.js and which Linux distributions
>   are supported and how to install it.
>   https://github.com/nodesource/distributions
> 
> 
>                           SCRIPT DEPRECATION WARNING

stephendade, 2023-09-15:
I've just tested this on a Pi ZeroW2 using RasPiOS Bullseye (11). I did get some warnings with ``npm install``, but no errors on the ``npm run build`` step.

This was using nodejs 20.6.1

Could you try again? I've just merged in a better nodejs install method.

ddd999, 2023-09-18:
I tried again with a (new) fresh install of Raspbian Lite, and it worked without errors. Thanks!

I did still get the "Possible EventEmitter memory leak detected." warning a bunch of times but it seemed to still complete the installation fine despite that.

closed 2023-09-18 by ddd999