stephendade/Rpanion-server #166: error

lin19968, 2023-10-05:
How to solve this problem
[image]

lin19968, 2023-10-05:
When I use the serial stream played by VLC, it stops playing for 2 seconds, how can I solve this?

lin19968, 2023-10-05:
I'm using Ubuntu 22.04

stephendade, 2023-10-06:
The ``deprecated`` warnings are normal during installation. They don't affect the operation of Rpanion-server.

For the video, which camera and settings are you using?

lin19968, 2023-10-06:
[image]
I use this camera, and when I use the official Raspberry Pi image, I can output video normally, but I can't configure it myself

stephendade, 2023-10-06:
I need the other settings on the page too. Also the actual name of the camera.

>  I use this camera, and when I use the official Raspberry Pi image, I can output video normally, but I can't configure it myself
I'm not sure what you mean by "can't configure it myself"

lin19968, 2023-10-06:
[image]
I use Ubuntu 22.04 system to run Rpanion-server, I can't output video, I burn the official configuration Raspberry Pi image to output video, I don't know what the problem is.

lin19968, 2023-10-06:
[image]
This is the packaging of the camera

stephendade, 2023-10-06:
I've confirmed the freezing happen when using RTSP. Can you test RTP too - that works ok for me.

lin19968, 2023-10-06:
It is also impossible to output video using RTP

stephendade, 2023-10-08:
> I use Ubuntu 22.04 system to run Rpanion-server, I can't output video, I burn the official configuration Raspberry Pi image to output video, I don't know what the problem is.

I've had a chance to run some tests with an MJPEG source camera.
- Ubuntu22 is quite unreliable with streaming. I don't recommend you use it.
- I had no issues using a recent version of RasPiOS, or the 0.1.0 image in the "releases" tab.

stephendade, 2023-10-08:
Update: I've added a patch to make streaming more stable on Ubuntu22 - I changed to software x264 encoding and it works much better.

lin19968, 2023-10-09:
Now that the problem has been solved, thank you very much

lin19968, 2023-10-09:
I tried to install the Ubuntu 20.04 desktop version of the system, but I couldn't find the corresponding .img file online, so I had to choose to install the 22 system

closed 2024-10-10 by stephendade