stephendade/Rpanion-server #169: Feature Request: send VIDEO_STREAM_INFO mavlink messages to GCS

rmackay9, 2023-10-13:
It would be convenient for the user if RPanion sent out [VIDEO_STREAM_INFORMATION messages](https://discuss.ardupilot.org/t/rpanion-server-0-10-released/107191) messages to the GCS so that it could more easily find video streams and pop-up a window for the user.  As far as I know, none of the GCSs will consume this message yet but I think they probably should.

The related [MP enhancement request is here](https://github.com/ArduPilot/MissionPlanner/issues/2906).
The related [QGC PR is here](https://github.com/mavlink/qgroundcontrol/pull/10667).

This request was first raised on [this AP blog post](https://discuss.ardupilot.org/t/rpanion-server-0-10-released/107191).

stephendade, 2023-10-14:
That seems do-able. MAVLink definitions are here: https://mavlink.io/en/messages/common.html#VIDEO_STREAM_INFORMATION.

From what I can see, this message is in response to a ``MAV_CMD_REQUEST_MESSAGE`` and is not streamed by default? Also should I be streaming the message or just sending a single one?

rmackay9, 2023-10-15:
Hi @stephendade,

Thanks.  Just sending it in response to MAV_CMD_REQUEST_MESSAGE would be OK I think.

ddd999, 2023-11-12:
@stephendade I'm starting work to implement this as it ties in to #167 as mentioned.

Is there a simple way to send custom MAVLink messages (like MAV_CMD_REQUEST_MESSAGE and the other camera-related messages) for testing?

Mission Planner only has a few messages available by default. I don't see a way to add any to the "Actions" or "Simple Actions" tabs short of re-building MP.

Side note, I've followed the instructions on GitHub to rebuild MP and it builds successfully, but with a huge number of warnings. It then fails to run because of manifest definitions not matching assembly references (even after copying all the DLLs from the standard MP install).

rmackay9, 2023-11-12:
MP doesn't support any way to send custom messages but for the specific case of REQUEST_MESSAGE there is the message interval window which is hidden on the Ctrl-F page.
[image]

For sending custom messages I normally use MAVProxy which does run on windows.

MP's camera and gimbal controls are listed on these two pages:

- https://ardupilot.org/copter/docs/common-mount-targeting.html
- https://ardupilot.org/copter/docs/common-camera-controls.html

ddd999, 2024-03-13:
@rmackay9, VIDEO_STREAM_INFORMATION reply has been implemented. I guess there is still some work left to implement the rest of the camera protocol described here: https://mavlink.io/en/services/camera.html#video_streaming

Specifically:

- Advertise the camera device with a new heartbeat message
- Respond to requests for CAMERA_INFORMATION, including sending CAMERA_CAP_FLAGS_HAS_VIDEO_STREAM (to be modified in future when image and video capture are implemented)

Did I miss anything?