stephendade/Rpanion-server #175: No video feed thru UDP

ltrzcinski, 2023-10-19:
There is no video feed thru UDP when configured as follows:

(RTSP works very good)

[image]

bbc25185, 2023-10-20:
I see Destination IP 127.0.0.1
You need to change the IP address from 127.0.0.1 (its the IP of LocalHost) to the IP of the machine where QGC run.

lida2003, 2024-01-28:
> You need to change the IP address from 127.0.0.1 (its the IP of LocalHost) to the IP of the machine where QGC run.

I think it's [ Feature Request: send VIDEO_STREAM_INFO mavlink messages to GCS #169 ](https://github.com/stephendade/Rpanion-server/issues/169).

closed 2024-10-10 by stephendade

Navneet2103, 2025-05-14:
my feed comes at low latency but grey in colour! 

[image]

stephendade, 2025-05-14:
> my feed comes at low latency but grey in colour!

That's due to H264 compression artifacts. You'll need to wait ~30sec for the video stream to stabilise.