stephendade/Rpanion-server #176: Raspberry PI 4B failed to identify the flight control serial port when rpanion-server was installed in ubuntu22.04

Guangday, 2023-10-26:
Hi Stephen Dade, I followed the instructions to install rpanion-server and used automatic installation (Raspberry PI). No error is reported during the installation, and the installation succeeds. I logged on the port 192.168.0.100:3000 on the other PC, and I could get the video information correctly in the video stream, but there was no information loading at the flight control acquisition serial port. As shown in the following picture. I tried to use 'sudo chmod 777 /dev/ttyusb0' to increase the ttyUSB0 permission, but still couldn't get it. I'm sure my Raspberry PI is good, and directly burning the image file can run rpanion-server normally, I need your help!

Guangday, 2023-10-26:
[image]

Guangday, 2023-10-26:
How Raspberry PI connects to Flight control.
[image]

Guangday, 2023-10-26:
Hi Stephen Dade, let me download the error log of the server. The problem may be that the cmdline file of the Raspberry PI system is in the firmware directory of the boot directory, so the client cannot recognize the error, I don't know if my understanding is correct. I used the official website downloader for my Raspberry PI 4B and downloaded the ubuntu22.04LTS-desktop (64) direct installation version.

Guangday, 2023-10-26:
[image]

stephendade, 2023-10-26:
Thanks for the report - I'll investigate and get back to you.

stephendade, 2023-10-27:
I've fixed this issue in #177. Could you confirm this solves your issue?

Guangday, 2023-10-28:
Hi Stephen Dade, thank you for your help. I have solved the server connection on my Raspberry PI 4B. I hope you can make it more perfect!

closed 2023-10-28 by Guangday