stephendade/Rpanion-server #179: `pip install`s in deploy/install scripts fail when running on Raspberry Pi OS Bookworm

daphil19, 2023-11-09:
Starting with Raspberry Pi OS Bookworm, The system installed python adopted [PEP 668](https://peps.python.org/pep-0668/). This prevents running `pip install`s at the system level (even when using `--user`). The (recommended) alternatives include either installing system-level packages with `apt` or using a `venv`. The RPI [docs](https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi) recommend using `apt` where possible.

An example of the error (including the first line when running `./RasPi2-3-4-deploy.sh` can be found below:
```
+ sudo python3 -m pip install --upgrade pip
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

I'm not sure what other OS-es would be impacted by this, as I think it is up to the distribution to ship python binaries that enforce the PEP.

The workaround is to replace many of the `pip install`s with the appropriate `apt install`, but it is also necessary (in the case of `pymavlink` to use the new `--break-system-packages` flag.

stephendade, 2023-11-10:
Thanks for reporting! I've not had a chance to test with Bookworm yet - hopefully in the next few weeks

stephendade, 2023-11-29:
Merged and fixed

closed 2023-11-29 by stephendade