stephendade/Rpanion-server #180: RTSP not working on RPI OS Bookworm

daphil19, 2023-11-11:
I'm unable to get the RTSP stream to work on my Pi 4 running RPI OS Bookworm. The behavior I'm seeing is the client freezing after getting what appears to be the first frame of the stream, and then either crashing or remaining frozen (depending on the particular client in use).

I've tried both the test source as well as the USB webcam I'm currently using with vlc, `ffplay`, and GStreamer as clients.

The RTP/UDP stream works for me without issue. I'm also able to get a working RTSP stream using [`go2rtc`](https://github.com/AlexxIT/go2rtc), FWIW.

I'm not sure if this is any help, but here are the logs from `journalctl` from one of my attempts:

```
rpanion npm[4630]: GST stderr: 0:00:28.976760224  5034 0xf4a05418 WARN                    v4l2 gstv4l2object.c:4675:gst_v4l2_object_probe_caps:<v4l2h264enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument
rpanion npm[4630]: GST stderr: 0:00:28.979536476  5034 0xf4a05ab0 FIXME                default gstutils.c:4031:gst_pad_create_stream_id_internal:<videotestsrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
rpanion npm[4630]: GST stderr: 0:00:28.988472390  5034 0xf4a05ab0 WARN                    v4l2 gstv4l2object.c:3228:gst_v4l2_object_reset_compose_region:<v4l2h264enc0:src> Failed to get default compose rectangle with VIDIOC_G_SELECTION: Invalid argument
rpanion npm[4630]: GST stderr: 0:00:28.990503510  5034 0xf4a05ab0 WARN                    v4l2 gstv4l2object.c:3228:gst_v4l2_object_reset_compose_region:<v4l2h264enc0:sink> Failed to get default compose rectangle with VIDIOC_G_SELECTION: Invalid argument
rpanion npm[4630]: GST stderr: 0:00:29.053652454  5034 0xf4a05ab0 WARN          v4l2bufferpool gstv4l2bufferpool.c:848:gst_v4l2_buffer_pool_start:<v4l2h264enc0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
rpanion npm[4630]: GST stderr: 0:00:29.088566764  5034 0xf4a05a78 WARN              rtpsession gstrtpsession.c:2435:gst_rtp_session_chain_send_rtp_common:<rtpsession0> Can't determine running time for this packet without knowing configured latency
rpanion npm[4630]: 0:00:29.088644140  5034 0xf4a05a78 WARN              rtpsession gstrtpsession.c:2514:gst_rtp_session_chain_send_rtp_common:<rtpsession0> Don't have a clock yet and can't determine NTP time for this packet
rpanion npm[4630]: GST stderr: 0:00:29.090341308  5034  0x1390d28 FIXME              rtspmedia rtsp-media.c:4624:gst_rtsp_media_suspend: suspend for dynamic pipelines needs fixing
rpanion npm[4630]: GST stderr: 0:00:29.093488848  5034  0x1390d28 FIXME              rtspmedia rtsp-media.c:4624:gst_rtsp_media_suspend: suspend for dynamic pipelines needs fixing
rpanion npm[4630]: 0:00:29.093569317  5034  0x1390d28 WARN               rtspmedia rtsp-media.c:4663:gst_rtsp_media_suspend: media 0xf4a4c1c0 was not prepared
rpanion npm[4630]: GST stderr: 0:00:29.108513064  5034  0x1390d28 WARN              rtspstream rtsp-stream.c:5606:gst_rtsp_stream_query_position:<GstRTSPStream@0xf4a50228> Couldn't obtain position: position query failed
rpanion npm[4630]: GST stderr: 0:00:40.638236338  5034  0x1390d28 WARN               rtspmedia rtsp-media.c:4975:gst_rtsp_media_set_state: media 0xf4a4c1c0 was not prepared```

stephendade, 2023-11-23:
Just confirming that I can replicate this issue.

It's a very odd one. The RTSP stream will work fine if a use the x264 (software) encoder, but not the hardware (v4l2h264enc) encoder.

And both work on RTP!

I'm thinking there's some weirdness happening with how v4l2h264enc formats the h264 packets into the rtsp stream. Will investigate further....

stephendade, 2023-11-29:
So, I wasn't able to get the hardware encoder working in the end. So I'll make it fallback to software encoding on Bookworm. Not ideal, but hopefully the RasPi people will fix the issue at some point.

Fix is in #182

stephendade, 2023-11-29:
Merged and fixed

closed 2023-11-29 by stephendade