stephendade/Rpanion-server #183: NIVIDA TX2 NX installation failed

Guangday, 2023-12-11:
An error occurred while installing rpanion-server into nivida tx2 nx. install_common_libraries.sh terminates running, as shown below.
Run：pip3 install --upgrade pymavlink --user --break-system-packages
Q：no such option: --break-system-packages

stephendade, 2023-12-11:
What Ubuntu version and pip version are you using?

Guangday, 2023-12-12:
Ubuntu 18.046LTS，pip 21.3.1;

stephendade, 2023-12-12:
I've confirmed the bug. Fix is in #184. Thanks for reporting!

> If I am trying to modify rpanion-server on ubuntu, is there any recommended engineering compiler?

Rpanion-server is written in nodejs, as such there's no compiler required. Just the nodejs runtime.

Guangday, 2023-12-12:
There is also a problem, I installed the system using the NIVIDA official SDKmaneger, and the following error occurs in the command sudo apt install -y nodejs. My system didn't change any of the source files, and there may be other people who made the same mistake as me.
nodejs: Dependency: libc6 (>= 2.28) But 2.27-3ubuntu1.6 is being installed
E: You can't fix the bug because you require certain packages to remain as they are, which breaks the dependencies between packages.

Guangday, 2023-12-12:
I have found the cause of the problem. It is indicated in the official website file that ubuntu18 no longer supports npm and later versions. I hope you can mark it.
[image]

stephendade, 2023-12-12:
Ok, looks like Ubuntu 18 supports nodejs16 at the latest. I'll edit the scripts to use v18 instead of v20 - now in #185

closed 2023-12-12 by Guangday