stephendade/Rpanion-server #191: Can't set "Maximum Bitrate" beyond 10MB on Raspberry Pi3B+

lida2003, 2024-01-14:
if I set Maximum Bitrate more than 10Mbps, such as 11Mbps. It will not take effect and blank the web page.

Tried `gpu_mem=196`, no luck!

>[image]

>[image]

stephendade, 2024-01-15:
That's deliberate - I'd set the max allowable bitrate to 10Mbps. I've now upped this to 60Mbps. Fixed in #193.

lida2003, 2024-01-15:
50M/30M/25M ==> VLC can't work
10M/20M ==> VLC works

[image]

stephendade, 2024-01-19:
> 50M/30M/25M ==> VLC can't work
10M/20M ==> VLC works

Yep, that's expected. The x264 hardware encoder on the Pi can do a maximum of 20Mbps.

The encoder is level 4 compliant, and according to https://blog.mediacoderhq.com/h264-profiles-and-levels/, that equates to 20Mbps.

lida2003, 2024-01-19:
I checked link, it seems level 4 can run over 20Mbps when in different profiles. Please correct me if I'm wrong.

stephendade, 2024-01-21:
In #195 I've updated to using the "high" profile, allowing for up to 50Mbps. Tested on a 3b+ and works well.

Note that the "bitrate" setting in the web GUI is a **maximum** number. The actual bitrate depends alot on the movement in the frame.

lida2003, 2024-01-21:
> In #195 I've updated to using the "high" profile, allowing for up to 50Mbps. Tested on a 3b+ and works well.

I would be nice to have 50Mbps.

stephendade, 2024-01-21:
> I would be nice to have 50Mbps.

I meant "up to and including 50Mbps"

closed 2024-01-28 by lida2003