stephendade/Rpanion-server #194: Re-install process will fail due to the existence of binary(node/npm) links

lida2003, 2024-01-15:
If delete node/npm in /usr/local/bin, then re-install process will proceed. Or it will fail due to file(node/npm) existence.

```
$ ./RasPiZero-deploy.sh
...
node-v16.19.1-linux-armv6l/bin/
node-v16.19.1-linux-armv6l/bin/node
node-v16.19.1-linux-armv6l/bin/npm
node-v16.19.1-linux-armv6l/bin/corepack
node-v16.19.1-linux-armv6l/bin/npx
+ sudo ln -s /usr/local/lib/nodejs/node-v16.19.1-linux-armv6l/bin/node /usr/local/bin
ln: failed to create symbolic link '/usr/local/bin/node': File exists

$ ls /usr/local/bin
meson  node  npm  pip  pip3  pip3.11  pip3.9
```

lida2003, 2024-01-19:
Raspberry Pi3B+ doesn't have this issue. Maybe PiZero is the time to phase out.

stephendade, 2024-01-19:
Thanks for all your testing. Agree that I'll put the PiZero(1) on the "not supported" list.

lida2003, 2024-01-19:
> Thanks for all your testing. Agree that I'll put the PiZero(1) on the "not supported" list.

I hope it will be supported.

stephendade, 2024-01-19:
The 0.8.0 version has a prebuilt image for the Zero. So you could use that.

The major issue is that the Zero uses the older armv6 CPU, whereas the Zero2 and Pi 2/3/4 are all armv7. A bunch of the nodejs libraries no longer provide builds for armv6.

I'm happy to take pull requests to fix Zero support, but it's not something I'll be able to focus on.

lida2003, 2024-01-20:
> The 0.8.0 version has a prebuilt image for the Zero. So you could use that.

Thanks, that's an alternative way of using Rpanion. And I have found that [opencv is based on armv7 by default](https://learnopencv.com/build-and-install-opencv-4-for-raspberry-pi/). So I'll start to use Pi3B+ and then switch Jetson, which can be more powerful.

> 
> The major issue is that the Zero uses the older armv6 CPU, whereas the Zero2 and Pi 2/3/4 are all armv7. A bunch of the nodejs libraries no longer provide builds for armv6.
> 
> I'm happy to take pull requests to fix Zero support, but it's not something I'll be able to focus on.

OK, Maybe someone has the knowledge and time can help.

closed 2024-04-10 by lida2003