stephendade/Rpanion-server #196: Feature Request: pan-tilt gimbal controls

lida2003, 2024-01-22:
I think a pan-tilt gimbal conrol is very thoughtful design. 

This is my video about this 2D gimbal, https://www.youtube.com/watch?v=omBjam52Hq0

So what I'm proposed here: is it possible for Rpanion to intergrate this feature? And then it might be controlled by MP or QGC.

- https://ardupilot.org/copter/docs/common-mount-targeting.html
- https://ardupilot.org/copter/docs/common-camera-controls.html

There is pan-tilt HAT board for RaspberryPi and jetson.

- 【Wiki】www.waveshare.net/wiki/Pan-Tilt_HAT 
- 【Github】https://github.com/waveshare/Pan-Tilt-HAT
- 【Aliexpress】https://www.aliexpress.com/i/2251832799696289.html?gatewayAdapt=4itemAdapt

stephendade, 2024-01-22:
For the pan/tilt, you connect the gimbal directly to the flight controller and control it from there. I'm not sure why you'd need to go via the RPi?

lida2003, 2024-01-22:
> For the pan/tilt, you connect the gimbal directly to the flight controller and control it from there. I'm not sure why you'd need to go via the RPi?

Currently, camera and [gimble](www.waveshare.net/wiki/Pan-Tilt_HAT) are all connected to Rpi. So I think it's rather simple for communication.

GCS ==> Rpi ==> pan/tilt

stephendade, 2024-01-23:
Using PWM servo pan/tilt with ArduPilot: https://ardupilot.org/copter/docs/common-camera-gimbal.html#common-camera-gimbal

I do plan to add in the ``VIDEO_STREAM_INFO`` MAVLink messages, as it should allow better GCS management of the camera video stream.

lida2003, 2024-01-23:
> Using PWM servo pan/tilt with ArduPilot: https://ardupilot.org/copter/docs/common-camera-gimbal.html#common-camera-gimbal
> 

Hmmmm... I see.

 Here is a scenario I'm considering:

a. pan/tilt camera, looking for a specific vechicle
b. now, the camera locks the vechicle at a pan/title angle
c. companion computer need to tell ardupilot to follow and tracking the vechicle
d. during the process of following and tracking, the camera should be always looking at vechicle

The quesiton is what kinds of MAVLink API or message should be used during the process? Any idea or tips?

stephendade, 2024-01-23:
So the camera is running some sort of image detection to find the car?

So a typical workflow would be:

1.  Camera detects the vehicle
2.  Camera software calculates relative pitch and yaw angles from drone to vehicle
3.  Camera software sends these values to ArduPilot via MAVLink message (https://ardupilot.org/dev/docs/mavlink-gimbal-mount.html#mav-cmd-do-gimbal-manager-pitchyaw-to-move-to-a-desired-angle-or-at-a-desired-rate)
4.  Repeat as needed

lida2003, 2024-01-28:
Ardupilot has this kind of features, which has more info about FC(navigation). Stick to the following diagram might be a good choice.

[image]

closed 2024-01-28 by lida2003