stephendade/Rpanion-server #197: Is there any way to save configuration locally on Rpanion?

lida2003, 2024-01-27:
I think it would be very nice if Rpanion can save the configuration, which was lastly executed sucessfully .

stephendade, 2024-01-29:
Thanks for checking. This feature _should_ work, but I've checked and it doesn't work anymore.

I'll investigate and fix.

stephendade, 2024-02-04:
#199 should fix the issue of the configuration not saving if streaming if active.

If streaming is not active, the configuration will not be saved.

lida2003, 2024-02-08:
> #199 should fix the issue of the configuration not saving if streaming if active.
> 
> If streaming is not active, the configuration will not be saved.

I don't think so, when I reboot from ssh, it didn't work the way as it supposed to do.

```
$ git log -n 1
commit f8766efd1ef0ff4af6fc9c1e6536e968ec73e4da (HEAD -> master, origin/master,                                                                                                                  origin/HEAD)
Author: Stephen Dade <stephen_dade@hotmail.com>
Date:   Sun Jan 21 14:43:44 2024 +1100

    Video: Latency improvements

```

[image]

lida2003, 2024-02-09:
Updated latest version, not good! Just check the below video.

 ```
$ git log -n 1
commit d066b49d67d32c3307be00f006e0ecaefcf93b19 (HEAD -> master, origin/master, origin/HEAD)
Author: Stephen Dade <stephen_dade@hotmail.com>
Date:   Mon Feb 5 16:09:05 2024 +1100

    Video: add correct values for testvideosrc and jetson CSI

```

https://github.com/stephendade/Rpanion-server/assets/1296171/8d06fd5c-7de5-49de-8c66-77b4823df300

stephendade, 2024-02-10:
Yep, I think my latest commits broke a few things ... I will investigate and get back to you.

stephendade, 2024-02-11:
It should be fixed in #202 (now merged into master). See if that works for you.

lida2003, 2024-02-11:
> It should be fixed in #202 (now merged into master). See if that works for you.

I works for me, Thanks.

[image]

closed 2024-02-11 by lida2003