stephendade/Rpanion-server #198: High CPU Usage on startup from NPM service

bbc25185, 2024-01-30:
I don't know if it's intentional or not, but when I turn it on I find the NPM service using the CPU a lot (screenshot attached) for about 5 minutes.
After 5 minutes the NPM service closes and CPU usage returns to normal levels (A few percent).
Can anyone explain to me why???
Thanks in advance for your answers.

Hardware:
Raspberry Zero 2
Rpanion version 0.10.0
Only configuration i touched: Connection to my WiFi Network

[image]

stephendade, 2024-01-31:
Thanks for reporting - I'll take a look in the next couple of days.

stephendade, 2024-02-04:
This is what I'm getting after 10min running latest master - no massive CPU usage of my Pi Zero2:

[image]

Could you try again with latest master and see if you still have the issue?

bbc25185, 2024-02-05:
Tried with latest master:
First 5 minute: Full cpu usage::
[image]
after 5 minute of uptime:
[image]

stephendade, 2024-02-07:
I'll need some logfiles to diagnose the issue further:
- ~/Rpanion-server/logs/app.log
- The output of ``journalctl -u rpanion.service -n 200``
- The output of ``sudo systemctl status rpanion.service``

bbc25185, 2024-02-07:
[Journalctl -u rpanion.service -n 200.txt](https://github.com/stephendade/Rpanion-server/files/14193927/Journalctl.-u.rpanion.service.-n.200.txt)
[systemctl status rpanion.service.txt](https://github.com/stephendade/Rpanion-server/files/14193928/systemctl.status.rpanion.service.txt)
[app.log.txt](https://github.com/stephendade/Rpanion-server/files/14193929/app.log.txt)

The requested 3 files!

stephendade, 2024-02-07:
It looks like there are some errors with the streaming video. If you don't stream video, do you get the same issue?

bbc25185, 2024-02-08:
Video streaming stopped. Same problem: CPU usage at 100% until the 5th minute.
logfiles attached:
[app.log no video.txt](https://github.com/stephendade/Rpanion-server/files/14210087/app.log.no.video.txt)
[systemctl status rpanion.service no video.txt](https://github.com/stephendade/Rpanion-server/files/14210089/systemctl.status.rpanion.service.no.video.txt)
[journal -u rpanion.service -n 200 no video.txt](https://github.com/stephendade/Rpanion-server/files/14210090/journal.-u.rpanion.service.-n.200.no.video.txt)

stephendade, 2024-02-09:
It's due to this : https://github.com/nodejs/node/issues/49937

A ``sudo rpi-update`` to upgrade the kernel will fix the issue.

bbc25185, 2024-02-09:
I confirm that updating the kernel solved the problem.
Thank you!

closed 2024-02-09 by bbc25185