stephendade/Rpanion-server #204: 720P@20FPS latency is obviously there, any idea?

lida2003, 2024-02-11:
640x480  latency is OK. maybe 200ms, it's several hundred milliseconds level.

When I switched to 720P, it's NOT good. What might be the issue?
a) [OV5647  CSI module](https://www.arducam.com/product/5mp-fisheye-camera-m8-mount-lens-raspberry-pi/) it seems to support 720p60?
b) Rpi3B+ system, it seems CPU is high, why?? GPU encorder not used?
c) Can't launch 720P RTSP server, with 40/50/60FPS!!!

https://github.com/stephendade/Rpanion-server/assets/1296171/afa51a5f-2f98-4e70-ab80-6b47d9307ee8

https://github.com/stephendade/Rpanion-server/assets/1296171/065909b1-f5be-4aab-8c1c-fde121caf66d

https://github.com/stephendade/Rpanion-server/assets/1296171/f2b6d2fc-9440-44f2-a935-c89d7b31cc66

I'm using Rpi3B+ and latested Rpanion version ==>Jetson  (wired network, not wifi)

```
 $ git log -n 1
commit 402a3454995e28a25d130d3afcb74d5ec3ad1b6d (HEAD -> master, origin/master, origin/HEAD)
Author: Stephen Dade <stephen_dade@hotmail.com>
Date:   Sun Feb 11 14:44:05 2024 +1100

    Video: Fix stream saving for CSI cameras

```

stephendade, 2024-02-11:
Can you post the output of ``libcamera-hello --list-cameras``? That will tell you the supported resolutions and framerates of the camera.

lida2003, 2024-02-12:
Thanks for you quick reply!

> Can you post the output of libcamera-hello --list-cameras? That will tell you the supported resolutions and framerates of the camera.

```
$ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

```

> At 720p/30fps I'm getting <10% CPU usage on my 3B+.

I tried 720P @ 15FPS 19000KPS, still got ~60% CPU.

It seems 720P @ 15FPS 19000KPS still did get encoder involved in my Pi3B+

> Note that above a bitrate of 20Mbps, Rpanion-server switches to the software H.264 encoder - that's why you're seeing the high CPU usage.

Can you guide me to set  GPU acceleration to level 4, so I can test without any other parameter changes.

Rpanion supports menu selection as below resolutions for CSI port camera(OV5460):
- 640x480
- 1280x720
- 1640x922
- 19201080

It seems that the libcamera says the driver supports:
- 640x480 @58.92 fps
- 1296x972 @43.25 fps
- 1920x1080 @30.62 fps
- 2592x1944 @15.63 fps

lida2003, 2024-02-12:
```
$ uname -a
Linux rpanion 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

It seems that if I set bitrate to 19000Kbps then it will use level4. But I didn't get CPU down.

- 640x480 @30FPS_19000Kbps ~40%CPU
- 640x480 @50FPS_19000Kbps ~100%CPU

https://github.com/stephendade/Rpanion-server/assets/1296171/bed22fe6-031d-40cc-84ef-1381cc0b13bd

https://github.com/stephendade/Rpanion-server/assets/1296171/8cdf5237-63fa-4e9d-b6a7-36788ba1e413

https://github.com/stephendade/Rpanion-server/blob/402a3454995e28a25d130d3afcb74d5ec3ad1b6d/python/rtsp-server.py#L75-L93

stephendade, 2024-02-12:
I've had another look.

I'm getting ~70% CPU usage at 640x480x50fps at 19000kbps.

To confirm, as long as you're running RasPiOS Bullseye (not Bookworm) you'll get GPU encoding.

The CPU usage is likely due to the video stream moving in and out of the GPU memory. I doubt I'd be able to reduce it any further.

lida2003, 2024-02-13:
> I've had another look.
> 
> I'm getting ~70% CPU usage at 640x480x50fps at 19000kbps.

OK

> 
> To confirm, as long as you're running RasPiOS Bullseye (not Bookworm) you'll get GPU encoding.
> 

Yeah, It seems GPU is working. I have used libcamera-vid to run rtsp(640x480 @50FPS, which takes about 50% CPU)

> The CPU usage is likely due to the video stream moving in and out of the GPU memory. I doubt I'd be able to reduce it any further.

Is it possible to support 1080P@30FPS?  as I did use libcamera-vid to run rtsp it seems CPU is also around 50%.

 
[Video: OV5647_libcamera_cvlc_1920x1080x30FPS](https://www.bilibili.com/video/BV1Kp421d787/)

stephendade, 2024-02-13:
> Is it possible to support 1080P@30FPS? as I did use libcamera-vid to run rtsp it seems CPU is also around 50%.

Tested here ... no issues running 1080p at 30FPS.

lida2003, 2024-02-13:
> > Is it possible to support 1080P@30FPS? as I did use libcamera-vid to run rtsp it seems CPU is also around 50%.
> 
> Tested here ... no issues running 1080p at 30FPS.

Latency is high, although CPU is NOT that high. Maybe Pi3B+ is just suitable for 640x480@30FPS

stephendade, 2024-02-14:
What sort of latency are you seeing? I'm getting ~0.5sec. RTP usually gives much lower latency (compared to RTSP), so you can try that too.

lida2003, 2024-02-15:
> What sort of latency are you seeing? I'm getting ~0.5sec. RTP usually gives much lower latency (compared to RTSP), so you can try that too.

Hoo.... Great RTP choice.  It's good enough now for simple video tests.  I can bring up bandwidth to 50000kbps, and actually it's around 25000bps 1920x1080@25FPS.

Thanks for the help!

[Jetson_Rpanion_RTP_1920x1080_30FPS](https://www.bilibili.com/video/BV1P6421g7Gd/)

closed 2024-02-15 by lida2003