stephendade/Rpanion-server #205: Error when setting AP Security to 'none'

chrissweeting, 2024-03-02:
I have a devices that for whatever reason will not connect to rpanion's AP running on a RPI Zero 2W. A possible solution is to set rpanion's AP security to 'none.' However when trying to set security to 'none.' The below error pops up.

Error editing network: Error: failed to modify 802-11-wireless-security.key-mgmt: ‘wpa-none’ not among [none, ieee8021x, wpa-psk, wpa-eap, wpa-eap-suite-b-192, sae, owe]

chrissweeting, 2024-03-04:
Setting security to 'none' now solves the previous error. However after setting security to 'none', the below error is given after trying to activate the AP. Also, the AP is non-functional until security is set back to wpa-psk, password is set, and given a reboot.

Error activating network: Warning: password for '802-11-wireless-security.wep-key0' not given in 'passwd-file' and nmcli cannot ask without '--ask' option. Error: Connection activation failed: Secrets were required, but not provided Hint: use 'journalctl -xe NM_CONNECTION=fb0e1680-694e-429a-b843-3298c5a6e082 + NM_DEVICE=wlan0' to get more details.

stephendade, 2024-03-05:
Ok, I think it was a missing "sudo" when bringing the connection up. I've now patched it, so see it that works for you.

chrissweeting, 2024-03-05:
Updated via the web tool and gave that a try. Upon reboot after security has been set to none, the AP is not active. Clicking on 'Activate' produces the same error.

chrissweeting, 2024-03-13:
I am using a RPi Zero 2 W and the versions below are what I'm seeing.

OS version: Raspbian GNU/Linux - 11
Node.js version: v20.7.0
Rpanion-server version: 0.10.0

stephendade, 2024-03-16:
This should be fixed, now that I've merged in #209. So give latest master a run and see if that works for you.

closed 2024-04-13 by stephendade