stephendade/Rpanion-server #207: npm install hangs at "fsevents" on Pi Zero 2W

ddd999, 2024-03-05:
I have tried this a few times with a fresh installation of Rpanion, i.e., after running:

`cd ~/ && git clone --recursive https://github.com/stephendade/Rpanion-server.git`
`cd ./deploy && ./RasPi2-3-4-5-deploy.sh`

Every time, the installer (running from a terminal via ssh) freezes at this state:

`(#########⠂⠂⠂⠂⠂⠂⠂⠂⠂) ⠋ reify:fsevents: sill reify mark deleted [ '/home/pi/Rpanion-server/node_modules/fsevents' ]`

Which seems to be because something being installed requires fsevents, but fsevents is specific to MacOS. See: https://github.com/angular/angular/issues/13935. The package.json for Rpanion doesn't require fsevents anywhere, so there is apparently a sub-dependency causing the problem.

In an x86 VM, the _npm install_ script hangs for quite a while (~30-60s maybe? I didn't time it) at this point as well, but is eventually able to complete and exit.

On the Pi Zero 2W, the whole system hangs-- I can't Ctrl-C out of the script, and if I try to open a new ssh connection there is no response. Power cycling is the only way I've been able to get out of this.

stephendade, 2024-03-05:
My understanding is that npm would do an OS check and not install fsevents on non-mac systems

Can you check your ``node_modules`` folder to see if fsevents is installed or not? A quick check on my x86 laptop reveals that fsevents isn't installed.

stephendade, 2024-03-05:
It might be a RAM issue. If the system has <1Gb RAM, then npm may have issues.

When installing on the Pi Zero (1), I did have to add some extra options during build to reduce memory usage for npm. Otherwise it would freeze the system.

Take a look at ``./deploy/build_rpanion.sh``, there's a section near the top that does a RAM size check for systems less than 500Mb. Maybe I need to up that to <1000Mb.

Try changing that check and see if that works for you?

ddd999, 2024-03-06:
fsevents is not installed in ~/Rpanion/node_modules

The same system freeze still happens if I change the memory size check to 1000:
`if [ $(free -m | awk '/^Mem:/{print $2}') -le 1000 ]; then`

`cat /proc/meminfo` reports: `MemTotal:         436980 kB`, so it was already less than the 500MB check anyway.

ddd999, 2024-03-06:
If I change "npm install" to "npm install --omit=optional", it gets further in the install process but still freezes when it gets to installing "@aashutoshrathi/word-wrap"

stephendade, 2024-03-11:
I've tried:
- Increasing swap memory
- increasing ``--max-old-space-size`` to 350

Still getting the install freeze around fsevents. Looking at ``htop``, there's very little CPU being used and 35% RAM used by npm.

Will investigate further...

stephendade, 2024-03-12:
> I've tried:
> 
>     * Increasing swap memory
> 
>     * increasing `--max-old-space-size` to 350
> 
> 
> Still getting the install freeze around fsevents. Looking at `htop`, there's very little CPU being used and 35% RAM used by npm.
> 
> Will investigate further...

Progress: the above steps, **plus** using Bullseye 32bit appear to reliably work.

So I'm putting the issue down to some sort of 64-bit issue on the Pi Zero2.

ddd999, 2024-03-12:
Thanks for investigating. I will try with that configuration next time I have a chance.

closed 2024-04-01 by stephendade