stephendade/Rpanion-server #208: Not enough RAM for building on Raspberry Pi 3B+

codeceptsDE, 2024-03-08:
I'm having trouble building on a Raspberry Pi 3B+, 1Gb RAM is insufficient to build mavlink-routerd.

I had to change `deploy/build_mavlinkrouter.sh` from 
```
if [ $(free -m | awk '/^Mem:/{print $2}') -le 500]; then
    ninja -j 2 -C build
else
    ninja -C build
fi
```
to `if [ $(free -m | awk '/^Mem:/{print $2}') -le 950 ]; then` to make it work. 
It's a bit slower, but at least it doesn't crash the Pi.

Maybe the same should apply to `deploy/build_rpanion.sh`, but I found it not to be necessary (right now). But it's cutting it close.

stephendade, 2024-03-12:
Thanks for the report. Happy to accept a PR with those changes.

timtuxworth, 2024-03-12:
Wouldn't just increasing the swap space work? The default is only 100k.

PR #212 (2024-04-01): Deploy: Increase swap size for Raspberry Pi builds
Raspbian sets the swap size to 100MB by default, which causes the npm install step to hang the system on devices with 500MB of RAM. About 700MB of total memory is needed for the build/install process.

This commit adds a step to build_rpanion.sh to increase the swap size, which enables the installation to complete successfully.

Fixes issue #207

Edit: Sounds like this may also fix #208?
closed 2024-04-01 by stephendade