stephendade/Rpanion-server #214: [Feature Request] IP bind to mac when Rpanion used as WiFi AP

lida2003, 2024-03-31:
As I'm using RTP, QGroundControl/MissionPlanner on mobile device will change if there are quite a lot of device in the network.

Is it possible to bind the IP of QGroundControl/MissionPlanner to MAC when Rpanion is used as a WiFi Access Point.

Or there is any other alternative way to use RTP without setting ip address on web panel?

PS: I noticed there is multicast RTP address, but I don't know how.

stephendade, 2024-04-10:
> Is it possible to bind the IP of QGroundControl/MissionPlanner to MAC when Rpanion is used as a WiFi Access Point.

There should be a setting on your laptop to specify a static IP address instead of using DHCP for the Rpanion network.

> If I set RTP to broadcast address 192.168.1.255, it seems RTP server is in infinite loop(restart), as the video is on/black/on/black....

Multicast RTP isn't supported at this time.

lida2003, 2024-04-10:
> There should be a setting on your device to specify a static IP address instead of using DHCP for the Rpanion network.

OK, it's a work around approach. And it's OK for me.

closed 2024-04-10 by lida2003