stephendade/Rpanion-server #215: Can't connect to FC through serial port

lida2003, 2024-04-01:
I don't know why pop up this error message? 

[image]

[image]

lida2003, 2024-04-01:
Hummmm.....

It seems that there is something wrong in file read/write permission???

[image]

lida2003, 2024-04-01:
a) Re-installed with latest git code, json error gone.
b) If I use an usb-serial adapter, it's OK to communicating with FC @ 57600bps
c) switched to Rpanion, I still didn't get FC connected. 

[image]

[image]

stephendade, 2024-04-10:
> c) switched to Rpanion, I still didn't get FC connected. (Rx/Tx switched useless)

Can you give a full list of all detected Serial devices?

lida2003, 2024-04-10:
> Can you give a full list of all detected Serial devices?

[image]

stephendade, 2024-04-10:
> BTW, I didn't find the device on dev dir.

Try ``ls /dev`` to get the full list

lida2003, 2024-04-10:
> > BTW, I didn't find the device on dev dir.
> 
> Try `ls /dev` to get the full list

[image]

I can get data from linux console, hardware should be OK

[image]

lida2003, 2024-04-10:
Configure as following steps:

1) start telemetry
[image]
2)configure  QGC
[image]
3) connect
[image]
4) check QGC telemetry status  ,,,,  "disconnect"???
[image]

lida2003, 2024-04-10:
I have got a video, it seems that  rpanion has recevied characters from UART, but it failed to parse.
https://github.com/stephendade/Rpanion-server/assets/1296171/483c64de-2893-4626-a7c8-fcb547d1dc28

stephendade, 2024-04-10:
Can you try ``/dev/serial0`` with MAVProxy on the RPi? This would confirm if the serial port is working properly or not.

lida2003, 2024-04-10:
> Can you try `/dev/serial0` with MAVProxy on the RPi? This would confirm if the serial port is working properly or not.

Yes, I have connected serial to computer, it works fine with mission planner.

stephendade, 2024-04-10:
Ok, so it's a configuration issue on the RPi?

What model RPi are you using?
Can you post the contents of your ``/boot/config.txt`` and ``/boot/cmdline.txt``
Maybe try connecting on ``dev/serial1`` using MAVProxy, just to confirm it's not a mixed up serial port?

lida2003, 2024-04-11:
Yeah, it Rpi serial config issue. And I'm using RPi 3B+ hardware.

After searching/reading links, I finally get it running.

A) cmdline.txt to remove "console" config.

```
$ cat /boot/cmdline.txt
root=PARTUUID=6ed124ff-02 rootfstype=ext4 fsck.repair=yes rootwait
```

B) Adding following lines to use the correct uart.

```
enable_uart=1
dtoverlay=uart1
dtoverlay=disable-bt
core_freq=250
dtoverlay=pi3-miniuart-bt
```

@stephendade althouhg it's running, I still have a few quesiton here.

a) I have looked into /boot, there is no dtd file called "pi3-miniuart-bt", quite confused about `dtoverlay=pi3-miniuart-bt`, Can I remove `dtoverlay=pi3-miniuart-bt`?
b) There are two ways for using Rpi serial: hardware/software UART. Software is related to `core_freq=250`, Can you help to let me configure to use hardware uart config?

stephendade, 2024-04-11:
The 3B+ can be a bit strange for UART config sometimes.

In theory, you should only need:

> A) cmdline.txt to remove "console" config.
> 
> $ cat /boot/cmdline.txt
> root=PARTUUID=6ed124ff-02 rootfstype=ext4 fsck.repair=yes rootwait

and

> B) Adding following lines to use the correct uart.
> 
> enable_uart=1

Otherwise, I might need to add a section in the documentation about the 3B+ "proper" config.

lida2003, 2024-04-11:
> The 3B+ can be a bit strange for UART config sometimes.
>
> In theory, you should only need:
>
>
> and
>
>
> Otherwise, I might need to add a section in the documentation about the 3B+ "proper" config.

RPi3B+ need to do more

- `sudo systemctl disable hciuart`
- add `enable_uart=1` and `core_freq=250`

And now, it's quite OK. 

https://www.bilibili.com/video/BV1uC41157q2/

closed 2024-04-11 by lida2003

lida2003, 2024-05-08:
@stephendade I strongly recommend to use USB2TTL device. As it's NOT stable to use RPi3B+ TTL

- Config USB2TTL
[image]

- Pi's TTL
[image]

stephendade, 2024-05-12:
I've just used the Rpanion-server 0.10.0 image on a 3B+, with a Cube Orange flight controller TELEM2 port connected to the GPIO UART on the Pi.

Running at 57600 baud, zero issues. Worked the first time.

So I'm not sure why you're having issues.

lida2003, 2024-05-12:
> 
> Running at 57600 baud, zero issues. Worked the first time.
> 
> So I'm not sure why you're having issues.
> 

Maybe 57600, I'm running 115200