stephendade/Rpanion-server #225: 4G LTE networks results in significant screen freezing and stuttering

lida2003, 2024-05-09:
The use of 4G LTE networks often results in significant screen freezing and stuttering. 

However, when connected via wired or Wi-Fi networks with a VPN, these issues do not occur. 

Are there any solutions available? FEC/QOS  ???? any configuration that can help?

lida2003, 2024-06-02:
Currently, I don't think it's packet loss or 4G LTE signal issue.

But I'm NOT sure what cause the video image has quite a lot of noise mosaic.

https://www.youtube.com/watch?v=Q-lAwk4nHVo

stephendade, 2024-06-02:
The noise mosiac you're seeing in H264 compression artefacts. Does increasing the bitrate work? Otherwise, I may need to tweak the encoder settings.

lida2003, 2024-06-02:
There is not much improvement changing bitrate from 1100kbps to 2000kpbs,  but latency improved.  And It didn't improve either from 720P to 1080P.

stephendade, 2024-06-05:
Can you try some higher bitrates - 5000 to 10000kbps?   I found that much of the noise disappears above 5000kbps.

Also, you may need to mount your camera on some anti-vibration mounts. On multirotors and fixed-wing aircraft, heavy vibration can mess with the H264 encoder.

stephendade, 2024-06-05:
On the RPi, run ``v4l2-ctl -d 11 --list-ctrls-menu`` to get the x264 hardware encoder settings.

Those settings then go into here: https://github.com/stephendade/Rpanion-server/blob/master/python/rtsp-server.py#L148

Note, I recently pushed a large patch to ``rtsp-server.py``, so you may need to ``git pull`` on your RPi.