stephendade/Rpanion-server #226: OV5640 720P@60FPS start failed, log attached

lida2003, 2024-05-15:
OV5647 manufacturer claims that it should support 720P@60FPS, but right now, I can't configure 60FPS.

I wanna know if it's possible to use 720P@60FPS for OV5647?

```
daniel@rpanion:~/Rpanion-server/logs $ cat app.log
2024-05-15 10:35:55 info [server/index.js]: NTRIP stopped
2024-05-15 10:35:55 info [server/index.js]: Express server is running on localhost:3000
2024-05-15 10:35:56 info [server/index.js]: Opening Link /dev/ttyUSB0 @ 230400, MAV v2
2024-05-15 10:35:56 info [server/index.js]: Opened Router
2024-05-15 10:36:03 info [server/index.js]: [{"value": "/base/soc/i2c0mux/i2c@1/ov5647@36", "label": "CSI Port Camera (ov5647)", "caps": [{"value": "1920x1080xx-raw", "label": "1920x1080", "height": 1080, "width": 1920, "format": "video/x-raw", "fpsmax": "30", "fps": []}, {"value": "1640x922xx-raw", "label": "1640x922", "height": 922, "width": 1640, "format": "video/x-raw", "fpsmax": "40", "fps": []}, {"value": "1280x720xx-raw", "label": "1280x720", "height": 720, "width": 1280, "format": "video/x-raw", "fpsmax": "60", "fps": []}, {"value": "640x480xx-raw", "label": "640x480", "height": 480, "width": 640, "format": "video/x-raw", "fpsmax": "90", "fps": []}]}, {"value": "testsrc", "label": "Test Source", "caps": [{"value": "1920x1080xx-raw", "label": "1920x1080", "height": 1080, "width": 1920, "format": "video/x-raw", "fpsmax": "30", "fps": []}, {"value": "1280x720xx-raw", "label": "1280x720", "height": 720, "width": 1280, "format": "video/x-raw", "fpsmax": "30", "fps": []}, {"value": "640x480xx-raw", "label": "640x480", "height": 480, "width": 640, "format": "video/x-raw", "fpsmax": "30", "fps": []}]}]

2024-05-15 10:36:03 info [server/index.js]: [object Object],[object Object]
2024-05-15 10:36:03 info [server/index.js]: Started Video Streaming of /base/soc/i2c0mux/i2c@1/ov5647@36
2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  0:00:00.393532291   746     0x2be98b30 WARN                    v4l2 gstv4l2object.c:4403:gst_v4l2_object_probe_caps:<v4l2h264enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  [0:00:44.543747325] [746]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  [0:00:44.603336440] [747]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  0:00:00.460185781   746     0x2be3dd20 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<libcamerasrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  [0:00:44.620563784] [752]  INFO Camera camera.cpp:1033 configuring streams: (0) 1280x720-NV12

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  [0:00:44.623523315] [747]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  0:00:00.478517031   746     0x2be3dd20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted

2024-05-15 10:36:20 error [server/index.js]: startStopStreaming() err  0:00:00.478967396   746     0x2be3dd20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted

2024-05-15 10:36:21 error [server/index.js]: startStopStreaming() err  0:00:00.920584948   746     0x2be3dd20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted
0:00:00.920880625   746     0x2be3dd20 WARN            libcamerasrc gstlibcamerasrc.cpp:310:processRequest:<libcamerasrc0> error: Internal data stream error.

2024-05-15 10:36:21 error [server/index.js]: startStopStreaming() err  0:00:00.920958125   746     0x2be3dd20 WARN            libcamerasrc gstlibcamerasrc.cpp:310:processRequest:<libcamerasrc0> error: streaming stopped, reason not-negotiated (-4)

2024-05-15 10:37:52 info [server/index.js]: Reset Video Settings
2024-05-15 10:37:52 info [server/index.js]: startStopStreaming() close null
2024-05-15 10:37:52 info [server/index.js]: Reset Video Settings

---- above stop video from web

2024-05-15 10:38:13 info [server/index.js]: Started Video Streaming of /base/soc/i2c0mux/i2c@1/ov5647@36
2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  0:00:00.141502495   937      0x11ce530 WARN                    v4l2 gstv4l2object.c:4403:gst_v4l2_object_probe_caps:<v4l2h264enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  [0:02:37.659326208] [937]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  [0:02:37.718334136] [938]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  0:00:00.207736207   937      0x1172d20 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<libcamerasrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  [0:02:37.734286995] [943]  INFO Camera camera.cpp:1033 configuring streams: (0) 1280x720-NV12

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  [0:02:37.735329124] [938]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  0:00:00.224667445   937      0x1172d20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  0:00:00.225194472   937      0x1172d20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted

2024-05-15 10:38:14 error [server/index.js]: startStopStreaming() err  0:00:00.670954539   937      0x1172d20 WARN                GST_CAPS gstpad.c:5701:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, colorimetry=(string)1:4:7:1, framerate=(fraction)30/1 not accepted
0:00:00.671242974   937      0x1172d20 WARN            libcamerasrc gstlibcamerasrc.cpp:310:processRequest:<libcamerasrc0> error: Internal data stream error.
0:00:00.671323234   937      0x1172d20 WARN            libcamerasrc gstlibcamerasrc.cpp:310:processRequest:<libcamerasrc0> error: streaming stopped, reason not-negotiated (-4)

---  above start video 720P 60FPS error
```

stephendade, 2024-05-15:
> I wanna know if it's possible to use 720P@60FPS for OV5647?

The camera is capable, but the hardware H264 encoder is limited to 30fps. You'd need to switch to using a software encoder instead.

lida2003, 2024-05-15:
> software encoder

Ah, I see.  720P@30FPS has the maxout the best performance of the Rpi3B+ .

closed 2024-05-15 by lida2003